canne/dashboard_tactics_pi #18: O-DT-TA-1-005 : To provide calculated Leeway data

canne, 2019-06-30:
The Leeway shall be calculated with minimum three input values, heel, speed on water and hullshape-factor.

tom-r, 2019-07-07:
Petri,
if you take the code from tactics_pi, this is fullfilled.
See page 6 of tactics_pi.pdf (Chapter 2.2):
_2.2 Calculate “Leeway”
Leeway describes the drift of the boat due to the force of the wind. Leeway is the basic 
input for the surface current calculation described later on. Input for the leeway calculation 
is your heel angle. Normally you'd say : the more you heel, the more you drift . But that's 
only part of the truth. Other significant inputs are boat speed and the shape of your hull…
A widely (NKE, B&G,...) used formula calculates the leeway with 3 input values : heel, boat
speed (STW), and hullshape-factor.
Leeway =   hullshape-factor*heel/(STW*STW)
To make this work, you have to estimate the hullshape-factor. The attribute in the preferences is called “Boat's Leeway factor [0..20]:”
The input range is 0...20, 10 is a good value to start with._

BR,
Thomas

canne, 2019-07-13:
Hi Thomas,
This is requirement capture (captured from your document, not rewriting or reinventing it). In a test plan an electronic and unique reference is needed to the requirement or requirements which needs to be fulfilled. This requirement, among others is referenced and tested in Tactics Leeway component https://canne.testquality.com/project/7362/testcase/177596/requirement/7349 (account is needed). Sorry, I am a system engineer...

closed 2019-08-10 by canne