canne/dashboard_tactics_pi #8: Add a watchdog in Dashboard for STW

canne, 2019-06-30:
Since the speed sensor paddle can get obstructed and stopped by the seaweed during the operations where _STC_STW provided speed values are used to TW calculations, all those would suddenly freeze. It is better that mStW in Tactics would go to NULL and so all values will disappear to alert the navigator. This needs to be done in Dashboard, with a timer. Of course, one could pop up a dialog and ask, do you want to swap to use SOG instead, but let's trust the skipper for now!

canne, 2019-06-30:
Implemented, see https://github.com/canne/dashboard_tactics_pi/commit/6d6a54894072673dd33606028c0952fed2792123#r34132451

closed 2019-06-30 by canne

canne, 2019-06-30:
Instead, created a sub-requirement in #15 of which this but one implementation. There can be others in the future, like a pop-up window, or such.

scadaperspective, 2019-07-04:
Something to add into watchdog IMHO ???

canne, 2019-07-04:
Ron, what function or action you would like the STW watchog to do when it triggers?

scadaperspective, 2019-07-04:
Sorry was a very glib reply.....

When I see people talk about alarms wakes the SCADA engineer in me :) 

same thing happens in audio plugins someone try to make the plugin do too may things some times.

Alarms, let watchdog do the alarm.

 As you probably know there is a plugin that handles "alarms" called watchdog it has a number of user configurable alarms based upon proximity or "health" of communications. It seems to work ok actually.

Like Plots - let the plugin in the suite that does that function do its job 

watchdog should in theory just need an additional code block for a new alarm condition added so it could be the log data comes into the API has a time stamp and quality assignment. The quality assignment to stop false positives needs some logic behind it perhaps ....

 I have done this on a PLC control system for merchant shipping... To verify a paddle wheel has not failed  found it was fairly reliable using an delta time increment counter  forward or reverse increment is valid, quality fail only if there is a delta change in position forward or aft movement from GPS position info compared to the log  movement and no pulse count happening = alarm condition you can adjust it to  prevent false positives ?

To me it is like trended data. 

let plots handle and store trended data. It could even be on a different pi box. 

Maybe this is all new API stuff. 
No the system doesn't need to be real time but at some point there needs to be a database for stuff that comes in from comms to be stored into and then plugins access the data. I guess I see bottle necks appearing and replication of efforts :)

ALL ok

canne, 2019-07-04:
Thanks Ron, I fully agree with you, the Watchog plugin shall centralize the alarms to the user. That's why I have not implemented but the silent action of "Dashboard watchdogs" - that's how these simple counters are called in Dashboard. For the discussion, we need to limit the scope to Dashboard Tactics and not to discuss about the system - albeit it would be interesting but there will be no stakeholders around from other sub-systems. Therefore, let's reformulate the issue:

Dashboard has several (five, exactly) counters (called Watchdogs) to change the state of its internal variables representing position, wind speed, etc. If a value gets an update, a corresponding watchdog is retriggered. If no updates (like connection loss) after ten seconds or so the counter hits zero and the "watchdog" is triggered, changing the state of the variable to "not valid". Soon after, the value disappears from the single line instrument or the dial drops to zero.

However, there is no counter (Dashboard watchdog) for STW. If Tactics has been set to calculate the true wind speed using the STW, it should be let known if the STW value is not valid anymore. Since the STW value is collected and communicated to Tactics by Dashboard, it is Dashboard's duty to inform Tactics about the eventual invalid STW data. I decided to implement that by adding a counter for Dashboard STW data, i.e. a "Dashboard STW Watchdog" which informs Tactics that STW data is not valid anymore if no new data is coming within ten seconds. This way, also Tactics stops producing true wind data when STW becomes invalid.

Seeing your description how the false positives etc. would require more careful implementation, I consider it unnecessary to add other features in this function (no pop-up windows or such).

Best regards,

Petri