canne/dashboard_tactics_pi #25: O-DT-TA-1-012 : Tactics instruments unique identification

canne, 2019-07-01:
Tactics shall identify itself with a leading ^ (caret "\u2038") when it creates an instrument showing active values such as Leeway using Dashboard instrument classes or classes which derives from those.

canne, 2019-07-03:
This discussion was originally started by @rgleason and @scadaperspective - please comment test and comment usability and ergonomics of the below solution, if you like, thank you!

In fact the ^  and unicode (caret "\u2038") are not the same, the latter is actually a tiny small character, barely visible. I replaced it with an up arrow \u2191. Details of the implemenation are here https://github.com/canne/dashboard_tactics_pi/commit/120bd102f9e7ee03b68ba34876fec7c9e34acfc9 . It is portable at least between Windows and Linux and renders like this:
[image]
and is displayed systematically on every instrument or function provided by Tactics. If no comments and/or a counter suggestion, I retain this solution after ten days from this message.

rgleason, 2019-07-04:
Great solution. Thank you Petri!

scadaperspective, 2019-07-04:
Up Arrow works on macOS

scadaperspective, 2019-07-07:
Cross platform to macOS is ok as well.

closed 2019-08-10 by canne