canne/dashboard_tactics_pi #34: incorrect Leeway direction with 'use heel sensor'

JoostFrouFrou, 2019-07-07:
when using my heel sensor from NKE the wrong Leeway is displayed (always to starboard). If I change the radio button to 'fixed heel' in the tab preferences the Leeway goes to port. 
[image]
[image]

attached a logfile with XDR to try
[NKE33935.LOG.pdf](https://github.com/canne/dashboard_tactics_pi/files/3365303/NKE33935.LOG.pdf)

canne, 2019-07-07:
Thanks, @joosfroufrou!  I'll have time to look at this in week or so. I will report back. @tom-r has published meanwhile v1.009 of Tactics here (layline improvements). Since I'm going to sync to changes in Dashboard-Tactics, of course, it would be good if you have time to test, meanwhile, again with this v1.009 version. That would be valuable information. You can report here, @tom-r has bee referenced and he can read this report (if he has time, he is also NKE-knowledgeable), no need for a doubles, your issue would be resolved quickest by concentrating the information about it. If you don't want to build yourself, you may need to wait a bit before @rgleason will make the v1.009 installer for Windows available in AppVeyor continuous integration platform.
Cheers,
Petri

canne, 2019-07-07:
Requirement specification #18.

tom-r, 2019-07-07:
Canne, Joosfroufrou,

I just ran the given sample file on original tactics_pi with O5. It works correctly.
You have wind from starboard, so boat heels and drifts to port ...
No change if I switch to fixed heel angle.
Trying one of my VDR files, with wind from port I heel and drift to starboard ...
Sorry, somehow I can't upload screenshots here ...
BR,
Thomas

PS: more than 30° of heel ? Could it be that you were a little bit overpowered that day ... ;-) ??

JoostFrouFrou, 2019-07-07:
Hi Thomas, yes heel was good 😁

If you look at my first screenshot you see that Leeway is wrong with

tom-r, 2019-07-07:
Joost,
I'm not really sure what your v1.009 of tactics_pi is and who created it ...
Until yesterday, the latest issue from me was v1.008, _yesterday_ I released v1.009 ...
Maybe Rick Gleason changed the rev to 1.009 as a O5 compile ...

Maybe the issue you found is just the false display of the ">" / "<" characters indicating the direction ...
Way in the past I introduced a "heel" instrument in dashboard, as well as True Wind direction.
I gave it the NKE style ">" for starboard, and "<" for port direction of Wind/heel
This worked for a while until someone else messed it up inadvertedly with a different change of dashboard first reversing them, then changing it to text (I think) ...
So forking tactics_pi from dashboard_pi, I corrected the false display in tactics_pi again ...
Maybe Petri now mixed the 2 characters up again while merging the code of both plugins, not knowing about the history of this ...
I'll talk to Petri on this ...

My NKE Regatta compass also supplies heel, but stops by design at 35° of heel.
I guess those armchair sailors designing the stuff can't imagine real life conditions ...  :-)

You're running Windows and O5, right ?
If you want I can email you the tactics_pi.dll I'm currently using (the real v1.009) ...
 
Thomas

JoostFrouFrou, 2019-07-07:
If you send IT to me I Will tryJoost

rgleason, 2019-07-07:
I have to look Thomas. I believe I have an ov50 release in my fork, and I believe that is what we are using for downloads. However I seem to recall that bdbcat also built it for ov50 early on. Mine was done by appveyor I believe.

rgleason, 2019-07-07:
I found an ov50 branch in my repos. There is an appveyer file under artifacts tab Click on the green check. I am cruising right now and using, cellphone

canne, 2019-07-07:
Hi Rick, I am travelling as well, but Thomas is not, so I think the best order to fix Jood issue caused by regrissve bug

rgleason, 2019-07-07:
Yes.

JoostFrouFrou, 2019-07-08:
The version I received from Tom solved the problem, good job.
I think it has to be changed in the new Dashboard_Tactics_pi to.

Joost

canne, 2019-07-08:
Thanks, Joost. I'll include the fix to the next version but that will take some time; I'll leave this issue open. Meanwhile, enjoy the Tactics Tom has fixed. Thank you both!
Petri

canne, 2019-07-08:
A personal memo from unicode.org for future works, please ignore: http://unicode.org/~asmus/ChartSymbolsInRunningText/ProposalPresentation.pdf
http://unicode.org/~asmus/ChartSymbolsInRunningText/ProposalPresentation.pdf
[image]

canne, 2019-07-12:
Implementation in dashboard_pi instrument, as defined by requirement #36 and sub-requirement #35 - implementation will require a modification to tactics_pi leeway data sentence data unit format.

canne, 2019-07-12:
This bug was caused because of me not going through the dashboard_pi XDR for Heel interpretation. "Heel" is dashboard_pi instrument, while "Leeway' is an active value which is using the Heel unit as default when a heel sensor is present. Since in the original tactics_pi the XDR sentence was modified to provide the string which, once passed to single line instrument provides the right symbol, that action was broken because it gets something like "on Port side" from dashboard after the degree sign. There was the same thing for the pitch sentence as well, so I fixed that as well, using tactics_pi type of nomenclature and thus allowing to meet the requirement #35 (arrow symbols, no text in English). The modified XDR sentence which is compatible both for dashboard_pi or tactics_pi is the following:

```
                    // XDR Pitch (=Nose up/down) or Heel (stb/port)
                    if (m_NMEA0183.Xdr.TransducerInfo[i].TransducerType == _T("A")) {
                        if (m_NMEA0183.Xdr.TransducerInfo[i].TransducerName == _T("PTCH")
                            || m_NMEA0183.Xdr.TransducerInfo[i].TransducerName == _T("PITCH")) {
                            if (m_NMEA0183.Xdr.TransducerInfo[i].MeasurementData > 0) {
#ifdef _TACTICSPI_H_
                               xdrunit = L"\u00B0u";
#else
                               xdrunit = _T("\u00B0 Nose up");
#endif // _TACTICSPI_H_                                
                            }
                            else if (m_NMEA0183.Xdr.TransducerInfo[i].MeasurementData < 0) {
#ifdef _TACTICSPI_H_
                                xdrunit = L"\u00B0d";
#else
                                xdrunit = _T("\u00B0 Nose down");
                                xdrdata *= -1;
#endif // _TACTICSPI_H_                                
                            }
                            else {
                                xdrunit = _T("\u00B0");
                            }
                            SendSentenceToAllInstruments(OCPN_DBP_STC_PITCH, xdrdata, xdrunit);
                        }
#ifdef _TACTICSPI_H_
                        else if ((m_NMEA0183.Xdr.TransducerInfo[i].TransducerName == _T("ROLL")) ||
                                 (m_NMEA0183.Xdr.TransducerInfo[i].TransducerName == _T("Heel Angle"))) {
#else
                        else if (m_NMEA0183.Xdr.TransducerInfo[i].TransducerName == _T("ROLL")) {
#endif // _TACTICSPI_H_
                            if (m_NMEA0183.Xdr.TransducerInfo[i].MeasurementData > 0) {
#ifdef _TACTICSPI_H_
                                xdrunit = L"\u00B0r";
#else
                                xdrunit = _T("\u00B0 to Starboard");
#endif // _TACTICSPI_H_
                            }
                            else if (m_NMEA0183.Xdr.TransducerInfo[i].MeasurementData < 0) {
#ifdef _TACTICSPI_H_
                                xdrunit = L"\u00B0l";
#else
                                xdrunit = _T("\u00B0 to Port");
                                xdrdata *= -1;
#endif // _TACTICSPI_H_
                            }
                            else {
                                xdrunit = _T("\u00B0");
                            }
                            SendSentenceToAllInstruments(OCPN_DBP_STC_HEEL, xdrdata, xdrunit);
                        }
                    }

```

canne, 2019-07-12:
Fixed by https://github.com/canne/dashboard_tactics_pi/commit/acceace7958eef81551f4546f201bf0b1459fc1b (build verified on Windows 10 and on Ubuntu 16.04 LTS not on Mac but that also builds in https://travis-ci.org/canne/dashboard_tactics_pi/builds/557961749 and at least up-arrows are known to render on it so let's wait for Mac users feedback for those three missing arrows I could not test on osx platform).
[image]
[image]

closed 2019-07-12 by canne

canne, 2019-07-13:
Confirmed with static test:
BitModern TestQuality Test Result https://canne.testquality.com/project/7362/plan/12830/run/104991/result/fullscreen/1287804

rgleason, 2019-07-14:
Canne , very interesting, careful documentation of testing.

canne, 2019-07-18:
`Created by TestQuality - Linked to TestQuality.com defect D10. Do not remove or edit.` [D10](https://canne.testquality.com/project/7362/result/1287802/defect/4569)

rgleason, 2019-07-18:
> 
> 
> A personal memo from unicode.org for future works, please ignore: http://unicode.org/~asmus/ChartSymbolsInRunningText/ProposalPresentation.pdf
> http://unicode.org/~asmus/ChartSymbolsInRunningText/ProposalPresentation.pdf
> [image]

Very interesting. Does it merit being linked in the Manual?

canne, 2019-07-19:
No: albeit the papers are a good reminder that the HMIs are moved / have to move to 21st century, they are just papers and can only perhaps used to feed the discussion why OpenCPN does not provide the plugins a purpose-made, cross-platform compatible icon library.

BTW, do you continue to receive e-mails from GitHub for this bug? It must be the canne.testquality.com site making modifications to tags when a regression test passes. Sorry about that; I can but suggest you to unsubscribe from this thread, there will but automated messages as far as I can see.

rgleason, 2019-07-19:
Comments continue to be sent via github. It is not a problem for me. Thanks.
I see many testquality tests!