canne/dashboard_tactics_pi #58: Polar::GetPolarSpeed() NaN check typoi at return

canne, 2019-07-27:
Instead of
`    return ((std::isnan(avspd1) || std::isnan(avspd1))?NAN: avspd1 + (avspd2 - avspd1)*fws);
`
it should read
`    return ((std::isnan(avspd1) || std::isnan(avspd2))?NAN: avspd1 + (avspd2 - avspd1)*fws);
`

canne, 2019-07-27:
Hi @tom-r, a minor typo I found while testing with Sean's example polars. If you like, I'll report also in your repo for your memo. Meanwhile, I continue tests with the change.

tom-r, 2019-07-27:
Thanks Petri,
good catch. Fixed it in the original tactics_pi repo.
Thomas

canne, 2019-07-27:
Created and linked to requirement #61 - preparation for fix commit., regression test reference [TC58](https://canne.testquality.com/project/7362/testcase/182178).

closed 2019-07-27 by canne