canne/dashboard_tactics_pi #59: Polar lookup table remains truncated with NaN lines below defined lowest TWA values

canne, 2019-07-27:
Visible easily with some very sparse polar files, like this one which comes with weather_routing_pi plugin:

[Example-15-30_pol.txt](https://github.com/canne/dashboard_tactics_pi/files/3438492/Example-15-30_pol.txt)

The resulting lookup table which Tactics build is not filled in below the 50 degrees of TWA with the averaged values:

[Example-15-30-out_issue-csv.txt](https://github.com/canne/dashboard_tactics_pi/files/3438495/Example-15-30-out_issue-csv.txt)

canne, 2019-07-27:
Observed the same behavior and symptoms in this test with tactics_pi v1.0.009:
[image]
It looks exactly the same with this polar from Sean as it does with my fork,

canne, 2019-07-27:
My analysis is that while the column zero is filled with 0 values (wind 0), that is not the case for row zero (angle 0) but NaN values are there in the BoatPolar->reset() method. I changed in that the following:

```
	for (int i = 1; i <= WINDSPEED; i++) {
		for (int n = 0; n < WINDDIR; n++) {
			windsp[i].winddir[n] = NAN;
			windsp[i].isfix[n] = false;
		}
```
to

```
	for (int i = 1; i <= WINDSPEED; i++) {
                windsp[i].winddir[0] = 0;
                windsp[1].isfix[0] = false;
		for (int n = 1; n < WINDDIR; n++) {
			windsp[i].winddir[n] = NAN;
			windsp[i].isfix[n] = false;
		}
```
and I got the above playback like this (I had to lower the speed to 2,7 knots to have reasonable polar performance with this example polar):
[image]

canne, 2019-07-27:
Hi @tom-r, here's another in the Polar-class, not urgent and perhaps not necessary to implement: if one has a well formed polar file the issue will not show up. But I tried with some (very limited) examples from weather_plugin_pi because I thought somebody without a polar file will try them out. If they will, the averaging based filling of non-defined values are likely to fail for lower wind angles. The culprit is in the class' reset() method (details in the report). I tested with tactics_pi v1.0.009 and it was the same error. I fixed it in my repo and now it works even with Sean's test polars.

I saw that you are implementing the recording functions now - I will wait for a new release before importing your new features since I am in the middle of the testing and boy that is time consuming! Hope it helps :grin:

tom-r, 2019-07-27:
Thanks Petri,
great job. I updated my local repo and will push in a few minutes.
Yes hold on please, I will add 2 boolean (0/1) opencpn.ini file keys to add clockticks & UTC time (in a later release then with ticks the GUI) for the exports.  I'll be out sailing for another week starting on August 10.
And I want to have something in the code to play with the basics of the polar creation/update...
Thomas

canne, 2019-07-27:
In in preparation for bug fix commit created and linked to requirements #60, #61 and #62 (the #62 covers, BTW, also creation of polar files from scratch - thinking about future enhancements).

Regression test created with reference [TC58](https://canne.testquality.com/project/7362/testcase/182178).

closed 2019-07-27 by canne

canne, 2019-07-30:
Sorry, @tom-r , my bad for not having time - or just being too lazy -  to copy and test my suggestions in your forked tactics_pi repo, from which I could make pull requests... The class based testing and comparing is a bit error prone. Now I am importing your latest changes with ediff and I see that there still is an error in your repo, in Polar::reset():
[image]