canne/dashboard_tactics_pi #63: Polar lookup table export has extra 2nd column separator

canne, 2019-07-28:
Expect in the first line, all lines that follow, have their second column separated by two tabs, which makes a CSV-import, for example to shift all column one step to the right.
[issue63before.zip](https://github.com/canne/dashboard_tactics_pi/files/3439281/issue63before.zip)

canne, 2019-07-28:
In _Polar::saveLookupTable()_ I changed
```
for (int n = 0; n < WINDDIR; n++){
            str = wxString::Format(_T("%d\t"), n);

```
to
```
for (int n = 0; n < WINDDIR; n++){
            str = wxString::Format(_T("%d"), n);

```
This fixed the issue:
[Uploading issue63after.zip…]()

canne, 2019-07-28:
Requirement mapping to #62

canne, 2019-07-28:
Hi @tom-r, here's one more in Polar:: not a big deal but if you plan to export files (as I suggested) in CSV format, please know that there is an extra tab which makes column 1 to skip. Fixed in my side to be able to complete the test (which revealed this issue).

closed 2019-07-28 by canne

tom-r, 2019-07-29:
Thanks Petri,

I've already fixed this a couple of days ago, when I started to play with the polar creation/update.
It's just not committed yet to Github, as I changed more than only this bit, which is still in work
Problem with csv is that the separator changes with your locale..
In Europe it's ';' (semicolon), in the US it is ',' (comma)...
But the O/QtVlm/... polar format uses ';', so I think it is safe here.
Thomas