canne/dashboard_tactics_pi #69: Baro history and True Wind history did not pass 4+ hour test

canne, 2019-08-04:
In the test, the three history graphs wind_history, polar_history and baro_history were executed with a recorded NMEA data for 4 hours minimum so that wind_history and polar_history were  also saving the data at their maximum speed at one second interval but the baro_history, back now with the 5 second timer tick was only saving every 60 seconds the data coming approximately once per minute. The graphical result is below:
[image]
We can observe that

- polar_perfomance seems to do fine
- wind_history has stopped after about 1 1/2 hours of operation
- baro_history has had some issues starting from 1 1/2 hours of operation but seems to do something still

The log files reveal the following:

**polar_performance**

```
Date;Time;AvgTWA;AvgTWS;smoothed BoatSpd;Percent
8/3/2019;10:28:34 PM;152;22.1;6.84;92.71
8/3/2019;10:28:35 PM;152;22.1;6.84;92.71
8/3/2019;10:28:36 PM;152;22.1;6.84;92.71
8/3/2019;10:28:37 PM;151;22.2;6.84;92.26
```
Starts OK but there are missing ticks and one second saving is not always kept:

```
8/4/2019;2:41:57 AM;163;21.3;6.89;106.14
8/4/2019;2:41:59 AM;162;21.2;6.90;106.65
8/4/2019;2:42:00 AM;162;21.2;6.92;107.14
8/4/2019;2:42:02 AM;162;21.1;6.94;107.61
```

**wind_history**

Looks alright (no good analysis for missing ticks, though), until about 1  1/2 of operation after which it goes all nuts:
```
8/4/2019;12:12:10 AM; -8;23.5;-12;22.2
8/4/2019;12:12:11 AM; -8;23.5;-12;22.2
8/4/2019;12:12:12 AM; -8;25.1;-12;22.2
8/4/2019;3:10:27 AM; -3;27.7;-12;22.3
8/4/2019;3:10:29 AM; -3;27.7;-12;22.3
8/4/2019;3:10:32 AM; -3;27.7;-11;22.4
8/4/2019;3:10:38 AM; -3;27.7;-11;22.4
8/4/2019;3:10:41 AM; -3;27.7;-11;22.5
8/4/2019;3:10:42 AM; -3;27.7;-11;22.5
8/4/2019;3:10:44 AM; -3;27.7;-11;22.6
8/4/2019;3:10:48 AM; -3;27.7;-11;22.6
8/4/2019;3:10:51 AM; -3;27.7;-11;22.7
8/4/2019;3:11:05 AM; -3;27.7;-11;22.7
```

**baro_history**

Remind this is now ticking back to 5 second interval timer, just that it started with the nearest ten second  moment. But it goes nuts what comes to data storage from the beginning, wonder is it the time stamp of the data which does not match. Anyway, it goes also mayhem after about 1 1/2 hour of operation.
```
Date;Time;Pressure
8/3/2019;10:32:00 PM;1018.3
8/3/2019;10:32:00 PM;1018.3
8/3/2019;11:07:00 PM;1018.3
8/3/2019;11:07:00 PM;1018.3
8/3/2019;11:07:00 PM;1018.3
8/3/2019;11:07:00 PM;1018.3
8/3/2019;11:24:00 PM;1018.4
8/3/2019;11:52:00 PM;1018.4
8/3/2019;11:52:00 PM;1018.4
8/3/2019;11:52:00 PM;1018.4
8/3/2019;11:52:00 PM;1018.4
8/3/2019;11:52:00 PM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
```
It is noteworthy that when closing down, the destructor (apparently) dumps a big number of lines like this (50, exactly):
```
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
8/4/2019;3:05:00 AM;1018.4
```

Please fix :grin: !

(data file) 
[nmeatest_bouches-r.zip](https://github.com/canne/dashboard_tactics_pi/files/3464784/nmeatest_bouches-r.zip)

canne, 2019-08-04:
@tom-r, FYI.
I've modified quite a lot the baro_history to get back the 5 second tick working, so I will concentrate on that now. Meanwhile, maybe you can look what can go wrong in true_wind history - since the polar_performance history works OK, thanks!

canne, 2019-08-04:
CPU usage was around 10 % during the test, all power saving features were disabled.

canne, 2019-08-04:
Compiled a version with a wxLogMessage to indicate if there is a wind unit change in wind_history, build a release. Small fix in baro_history to clear the m_Press while waiting on the slow barometers like mine. Shortened the test NMEA file 90 minutes from the beginning. Same setting as above, now the CPU load is 3 - 4% (previously I was running from the IDE). Let's wait and see.

canne, 2019-08-04:
After two hours of running, with the reduced data set, which is snipped approximately 90 minutes from its beginning [attached](https://github.com/canne/dashboard_tactics_pi/files/3464963/nmea-aller1_mistral_bouches_du_rhones_06.12._30kn_commence_-90minutes_apprx_from_beginning.zip)  I did not discover any issue with the wind_history and especially not at the moment when it crashed under the IDE. Looking in the opencpn.log file, my debug printing indicates that there has been issues with the VDR playback - timeouts for StW occur, for example.

I will test again this night with a full data set again, but without IDE.

Meanwhile, with or without the IDE, the baro_history looks odd what comes to timestamp interval (it has slowed down even further now after I added that m_Press = 0 in the tick). I suspect wxTimer, maybe I've done it wrong to move to 5s sampling.

So, for now **wind_history** OK but **baro_history** still issues.

canne, 2019-08-06:
<p>Tested successfully after a bug fix (frequency and barometer plot drawing)</p>

closed 2019-08-06 by canne

canne, 2019-08-06:
Test report
[2019-08-06_TCR406_longrun_plot_and_store_PASS.pdf](https://github.com/canne/dashboard_tactics_pi/files/3473710/2019-08-06_TCR406_longrun_plot_and_store_PASS.pdf)

End of test screenshot
[image]

End of test registered data
[tc49_csv_output_example.zip](https://github.com/canne/dashboard_tactics_pi/files/3473704/tc49_csv_output_example.zip)

rgleason, 2019-08-07:
Petri, impressive.