canne/dashboard_tactics_pi #72: Apparent Wind Angle

Nereus32, 2019-08-23:
Something odd...

Apparent wind angle is showing as a reciprocal of the actual apparent wind angle on MacOS Mojave, Ov5.01 and Dashboard_Tactics .3.05 & .40 (just upgraded). See screen shots. First is DBTactics showing the correct wind on compass rose and the reciprocal wind on the apparent wind gauge. I removed DBTactics and reenabled Dashboard, and all is fine (see screen shot 2). Also a screen shot of my iPhone showing the correct wind angle on the SeatalkNG network, so the data is right - matches the ST60 gauge on the boat. Is there a setting I set that has reversed the wind angle on the Apparent Wind Gauge in O5.01?

Steve

[image]
[image]
[image]

canne, 2019-08-23:
Hi, thank you for reporting and sorry for the bug I "introduced" in the "normal" (single needle) dial around v3.5. This has been reported already in bug #71 and fixed. There is an explanation how to fix this on your own (presuming you make your own build) in this [forum post](http://www.cruisersforum.com/forums/f134/tactics-plugin-166909-52.html#post2957880) (subject #1 addressed at the end of the post). Or you can take [this commit](https://github.com/canne/dashboard_tactics_pi/commit/9c37709e934ef47932d7833df83f3f9cf502bf4d) entirely, as it is. You can see from the continuous integration that it builds on osx. 

Note that I am developing in branch right now, awaiting also for a new feature from Tactics before I make a new release, which is a bit time consuming. It will take a couple of weeks, I reckon.

Sorry for the inconvenience, I have improved the regression testing of dials to make sure that this will not occur again.

canne, 2019-08-23:
Closing as duplicate of bug #71 .

closed 2019-08-23 by canne