canne/dashboard_tactics_pi #82: v0.4.1 Small issues

rgleason, 2019-09-27:
Petri please see.
http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087.html#post2984954

canne, 2019-09-28:
Hi Rick,

Thanks, looks nice; all efforts for simplification are welcome. I've finished the development of DashT with InfluxDB and SignalK plug-ins in a plug-in... The latter would allow turning the "Dash" part of the DashT to the motorboat skipper's taste. However, I'll continue only on data analysis and Tactics improvements after 0.5.2 which will be my last feature version, end of October.

rgleason, 2019-09-28:
Petri
Our efforts for simplification are still in a branch, but I've been focused on squiddio lately, and trying to rebase on mauro's master which now has the new plugin installer pr merged.

After the dust settles, I will be back to it.

canne, 2019-10-13:
This issue is relevant to original Dashboard code, incorporated in the same code base but not use in development. See answers

* http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087-35.html#post2985925
* http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087-36.html#post2991179

In the Dashboard-Tactics (_i.e._ this code base) the issue with jitter in data distribution is improved with a OpenCPN by-pass enhancement. There is a [report](https://github.com/canne/dashboard_tactics_pi/blob/a7bfc8dbe1563ba4b0eb7aa51c79de26df86689a/docs/signalk/analysis/Three-way_timestamps/Three-way_timestamps_win10.pdf) about the obtained stability improvement in the data distribution by separating it from the wxWidgets events framework.

canne, 2019-10-13:
Not relevant to the latest versions 0.5.x - the code based used is not Dashboard-Tactics but Dashboard only. For the time being it is not developed or maintained, only the Dashboard-Tactics code base. Therefore it is won't fix for this Dashoard-code-only report.

closed 2019-10-13 by canne