canne/dashboard_tactics_pi #90: Rudder Reversal?

rgleason, 2019-11-18:
Should be an easy one.
http://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html#post3018005

canne, 2019-11-18:
There is rudder angle in test use case [tc107](https://canne.testquality.com/project/7362/plan/12830/test/fullscreen/196044), see step 11 for bearing compass test. To be checked if the input file [tc107_RMB2.zip](https://github.com/canne/dashboard_tactics_pi/files/3856917/tc107_RMB2.zip)  (made from Thomas's boat recording test1.txt; I don't have rudder angle information back from Raymarine for some reason) has correct information concerning the rudder angle. I reckon I should make a specific test for this instrument. It should be possible by using the NMEA emulator, not a recording from a boat.

rgleason, 2019-11-18:
Petri
I used VDR_pi with Tactics Sample #2 file.
This file is in the opencpn manual for tactics and came be downloaded. I believe he has B&G instruments, so that was probably not a Raymarine Instrument.

Hope that helps.

canne, 2019-11-18:
Thanks, Rick, that's the same file we got, I just shortened it for the test use case. T. has NKE. I'll check with the emulator.

Nereus32, 2019-11-18:
Petri, 

I discovered this last night when I installed the rudder angle gauge. It works in Dashboard, but in Dash_T it's inverted. Hope this helps. BTW, I have a raymarine system, and I can make a VDR file for you with a bunch of rudder changes, if it helps.

Nereus32, 2019-11-18:
Here you go:

Obviously, I had to add a .pdf extension.

File is a recording of me turning to port (left) about 10 degrees, then more. Next I hard over to starboard and then make some more changes before coming back to center.

[Rudder Test.pdf](https://github.com/canne/dashboard_tactics_pi/files/3859548/Rudder.Test.pdf)

canne, 2019-11-18:
Thanks, that's very useful to have! I'll report here my findings soon.

rgleason, 2019-11-18:
Tried tc107_RMB2.txt  from Test Case 11
https://canne.testquality.com/project/7362/plan/12830/test/fullscreen/196044
I did not find a big enough turn to confirm operation of the rudder was right.

[tc107_RMB2.txt](https://github.com/canne/dashboard_tactics_pi/files/3860173/tc107_RMB2.txt)

This one has a hard right turn, to starboard just a ways in. You'll have to wait a while.
[Tactics-sample2-5m.txt](https://github.com/canne/dashboard_tactics_pi/files/3860176/Tactics-sample2-5m.txt)

canne, 2019-11-18:
I tested both with the emulator and your recorded data. The numerical instrument shows the right value, negative, as you said, first, "turn to port" and then positive, "turn to starboard".  It is the needle of the dial instrument which goes to the opposite, not very logical direction! Need to take a look why, since this code is not in Tactics and I just copied from the OpenCPN site from Dashboard. Some sign inversion along the road somewhere. Should not take too long. I'll improve the test use case to check also that the needle position makes sense and will make the (last I hope) release candidate 3 next weekend.

canne, 2019-11-20:
Tested both on out-of-box Dashboard w/ ov50 on Raspberry Pi 4B (4GB) and on a Dashboard compiled from the same code base as DashT but on Windows 10, both works as expected:
[image]
Same code base, different result? This is going to take some effort to debug, I am afraid, some derived class underneath has changed. Will be reporting back.

canne, 2019-11-21:
Fixed with 8dcf4b44b86c60145706b6458f7306efefe25b9e : Signal K provided timestamps allow individual timeout event handling for each instrument. This was not implemented in rudder dial instrument class while it was implemented in dial base class from which the rudder dial class derives from - this is mandatory feature in the bearingcompass and polarcompass and implemented by default in the base class.

As consequence, the `SetData()` method of the derived class did not match anymore the base class one and it did not actually override it, silently the method of the dial base class was called. But the base class method override was done precisely because the dial needle moves in rudder dial to the opposite direction with negative input values than in the base class dial. Hence the sign inversion never occurred and the dial base class needle moving rule prevailed.

BTW and for the record: if the timeout occurs, this instrument's default position in the constructor is set to 0. With this inverse logic, the timeout makes the needle to move to the zero position, which is the 50 degrees position on the right, that is in the green are. The numerical instrument shows dashes. Without SK timestamp, if the connection is lost, the needle remains static and so does the numerical instrument.

canne, 2019-11-21:
Tagged as regression since caused by a new function/feature.

canne, 2019-11-21:
Was not an easy one! Test report in https://github.com/canne/dashboard_tactics_pi/issues/94. Closing for now.

closed 2019-11-21 by canne

rgleason, 2019-11-21:
Nope, it wasn't. Pretty deep in there wasn't it? Thanks.