canne/dashboard_tactics_pi #94: Rudder dial instrument inversion in DashT

canne, 2019-11-21:
Fixed [https://github.com/canne/dashboard\_tactics\_pi/issues/90](<https://github.com/canne/dashboard_tactics_pi/issues/90>) with [https://github.com/canne/dashboard\_tactics\_pi/commit/8dcf4b44b86c60145706b6458f7306efefe25b9e](<https://github.com/canne/dashboard_tactics_pi/commit/8dcf4b44b86c60145706b6458f7306efefe25b9e>)

<br>

**Step 14 Pass**

**NMEA-0183 RSA**

Test with the NMEA simulator:

<br>

**Test without Signal K node server - Dashboard NMEA-0183 parser as reference**

<br>

Turn off Signal K node server you have been listening.

**Note**: Using this simulator, please take into account that it sends a sentence only for one rudder, while OpenCPN Dashboard requires a sentence for two rudders, even if the second rudder is not there. The two extra arguments can be easily added using NMEA Converter:

**$YDRSA,$GPRSA1,$GPRSA2,,V**

<br>

Rotate boat to port side by turning the emulator rudder slider to negative side, and observe the angle displayed being also negative:

[image]

<br>

<br>

Rotate boat to starboard side by turning the emulator rudder slider to positive side, and observe the angle displayed being also positive:

<br>

[image]

<br>

<br>

**Test with Signal K node server**

Provide two path of data from Signal K node server to OpenCPN as usual: the NMEA-0183 stream as TCP or UDP and activate DashT Signal K Input Streamer.

The input streamer interprets also the rudder angle from the emulators NMEA-0183 stream and sends it as Signal K data delta.

The purpose of this test is to verify that the displayed angles are consistent with the Dashboard's integrated NMEA-0183 parser.

**Note**: Signal K Node server does not need the extra rudder complying NMEA-0183 sentence as the Dashboard integrated NMEA-0183 parser does. Therefore you can use the simulator RSA signal with Signal K node server as it is. It comes well formed down to OpenCPN in both channels. If the NMEA Converter is enabled, disable it (not that it would have any effect to the below test but good practice requires that it is stopped).

<br>

[image]

[image]

<br>

Stop now the Signal K In streaming "instrument" (remove it). The operation returns now to the Dashboard integrated NMEA-0183 parser after approximately five seconds timeout.

<br>

**Repeat the test for the third time, now with the NMEA-0183 data which is coming from Signal K node server**.

<br>

<br>

<br>

canne, 2019-11-21:
https://canne.testquality.com defect [D33](https://canne.testquality.com/project/7362/result/1674228/defect/5900). Do not remove or edit.

canne, 2019-11-21:
Test report for https://github.com/canne/dashboard_tactics_pi/issues/90. Closing.

closed 2019-11-21 by canne

SCarns, 2019-11-22:
Great work. I can confirm it is correct on my Raymarine system.