canne/dashboard_tactics_pi #97: Installer tries to create confusing shortcut

transmitterdan, 2019-11-24:
The NSIS installer under Windows tries to create a shortcut in start menu folder "objsearch_pi".  I don't know from where it gets this folder name.  But there isn't any need for a shortcut with plugins so maybe we should change the installer to not try to install any shortcuts.

canne, 2019-11-24:
Thanks! I have no idea how this can be done, mine is copied from Sean. If you or somebody else knows how the NSIS installer internals work, a (tested) pull request is welcome. Otherwise, this issue is not a showstopper and can go to a maintenance patch later on.

canne, 2019-11-24:
For memo: in my Win10, can't find anything fishy in C:\Users\Petri\AppData\Roaming\Microsoft\Windows\Start Menu\Programs after the installation of rc3. I'll put the milestone in the next maintenance release, hopefully more information is available by then.

canne, 2019-11-24:
Here's a installer build from my machine, if somebody can test this (rc3) : [dashboard_tactics_pi-0.5.93-ov50-win32.zip](https://github.com/canne/dashboard_tactics_pi/files/3883824/dashboard_tactics_pi-0.5.93-ov50-win32.zip) we would get an idea is it NSIS on AppVeyor which is causing the issue?

transmitterdan, 2019-11-25:
That one does not try to create any shortcut.

canne, 2019-11-25:
On my Windows 10 which the above was built I have NSIS "nullsoft scriptable install system 3.0.4". Please report if this is an issue of the AppVeyor-built installation image or an issue with your own built environment.

transmitterdan, 2019-11-25:
It happened with my build environment. I have the same NSIS version as you do. I don’t know where it gets the obj search folder name.

closed 2019-11-25 by transmitterdan