canne/dashboard_tactics_pi #102: Should dashboard_tactics_pi use same configuration as old dashboard_pi?

transmitterdan, 2019-12-22:
In testing I have run into conflicts where this plugin “pollutes” the configuration settings of the old dashboard plugin. Probably because this is really just a branch of the old plugin but moved to its own repository. So many of the configuration parameters are no doubt spelled the same.

But is that really what we want? Don’t we want both dashboards to coexist peacefully?

canne, 2019-12-22:
It is, indeed, intentional: https://github.com/canne/dashboard_tactics_pi#configuration.

canne, 2019-12-22:
For the record, the captured main input requirement https://github.com/canne/dashboard_tactics_pi/issues/40 does not define explicitly this use case but the project title is quite clear regarding the scope: "_OpenCPN Dashboard plug-in with integrated Tactics instruments and functions_".

True. This **is** a fork of the OpenCPN v5.0 Dashboard. Plain copy of it. Enhanced with Tactics functions and instruments. Improved with new functions and bug fixes. It is intended to **replace** (in my boat at least) the distribution's Dashboard. Not to create yet-another-dashboard.

I have switched back and worth to original Dashboard and have not found any issue - apart a few dummy Tactics instruments if I have those, easy to remove even using the GUI. Nothing has been broken in the ini/conf file, just "polluted" (?).

This can develop into an additional requirement but [not before 2.0](https://github.com/canne/dashboard_tactics_pi/blob/c7656142b49c6bb38b3d1c681db9b2426f22193a/docs/developers/README.md#v1511-and-onwards).

transmitterdan, 2019-12-23:
I have found some issues when switching back and forth.  For example, if a polar is active in tactics version then switching back to the old dashboard_pi leaves a tiny x on the screen with no obvious way how to get rid of it.  I could imagine some other difficulties arising.

Really don't understand the logic in using the same configuration scope for two different plugins especially since the plugins do not implement the same instruments.  What is the benefit?

canne, 2019-12-23:
There is no benefit. If one is satisfied to the original ov50 Dashboard and 1.0.11 Tactics plug-ins one shall continue to use those and not use this plug-in.

If one wants to use this plug-in, one shall **not** use the original ov50 Dashboard. One may use the 1.0.11 Tactics plug-in (but there is no benefit doing so).

**This plug-in is a replacement of the ov50 Dashboard, not a new plug-in**. One needs to decide which of the two to use. Nothing is broken permanently though, if all of them are used simultaneously, but: what is the benefit? 

Like usual with **_O_**, if you just one to try out something, it is a good idea to make a copy of your .ini/.conf file to be able to roll back. Documentation on the front page / release page should suggest this, TBD.

Your tiny x under ov50 Dashboard are probably instrument window panes which have Tactics instruments allocated in them but ov50 Dashboard does not recognize them and the window pane remains totally empty, zero size. Go to Preferences, delete the Instrument panel with no or unknown instruments. Sorry for the inconvenience.

canne, 2020-05-17:
Fixed in v1.98.105 https://github.com/canne/dashboard_tactics_pi/commit/22b1410d0862a136ebbaf19f64b9927baa35f4a9 - no plans to jailbreak v1.5.11 branch, third party fork remains possible, of course. Closing for now.

commit 22b1410d (2020-05-17): refactor: v5 dashboard jailbreak ends (fixes #102)
closed 2020-05-17 by canne