canne/dashboard_tactics_pi #103: French translation issue

rgleason, 2020-01-17:
Schnapsy reports:
There is a problem with the translation (file *.po / *.mo ...).
In Dash-T, the French translation of the device names does not appear above the dials, it is the name in English that appears.
And if I modify the file .po /.mo (opencpn-dashboard_tactics_pi.mo), no change is taken into account...

canne, 2020-01-18:
Hi Rick, there is perhaps a misunderstanding - this is not a run-time feature but a build problem - thus depending of the developer's environment. By default _Dash T_ internationalization is not even possible in OpenCPN development framework since, following the [procedure for Plug-in Internationalization](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:internationalization:plugin_languages#plugin_languages), the plug-in must be official for that this works. One can find  [French translation from Crowdin](crowdin.com/project/opencpn/fr/) for _dashboard_pi_ and for _tactics_pi_ but **not** for _dashboard_tactics_pi_.

Of course, I will **not** attempt to somehow merge those two translations together, so until there is community effort to create translations for _dashboard_tactics_pi_ it will remain as it is.

<hr>

I'll keep this ticket open for eventual volunteers who are invited to read the above links for the instructions but also pay attention to this plug-in's [manifesto and roadmap](https://github.com/canne/dashboard_tactics_pi/tree/pr151/docs/developers), in order not to attack 1.5.11 translations without further ado.

<hr>

Back to Schnapsy's problem: If I remember correctly, he just uses my code base to have a convenient way to have a hackable, stand-alone _dashboard_pi_ ( **as it was** in ov50 - with conditional compilation ). Theoretically, it should work for him, since the plug-in name is now _dashboard_pi_. If it does not build for him _en français_, it is probably his environment (Windows, if I am not mistaken?).

After Windows build, find file (the path may change a bit for him):
```
C:\Users\Petri\opencpn\dashboard_pi\build\Win32\RelWithDebInfo\dashboard_pot-update\dashboard-pot-update.log
````
It usually give an idea what is wrong. For example, if his GNU gettext-installation is too old (we are following Sean's and Pavel's Weather Routing Plug-in's settings here, see `cmake/PluginLocalization` - they use some arguments for GNU `xgettext` which may not be present in Schnapsy's  installation).

This is easy to test, move to the project's root directory and issue command from the Windows command prompt:
```
"C:\Program Files (x86)\GnuWin32\bin\"xgettext.exe --force-po --package-name=dashboard_pi --package-version="1.5.11" --output=po/xxx.pot  --keyword=_ --width=80 --files-from=po/POTFILES.in
```
That's a command from the Cmake-file. One should find `xxx.pot` file now. If the command is complaining about unknown arguments, the the GNU gettext-package is too old. Get the latest (for Windows) from [here](https://mlocati.github.io/articles/gettext-iconv-windows.html) and try again. Until the above works, there is no reason to continue.

There maybe other problems caused by the conditional compilation (like with line numbers) but since I do not really see what might be the interest to keep up the old _dashboard_pi_ alive in my code base, I will mark this issue "Wont' fix" (and and I am not inviting pull-requests which fix _dashboard_pi_ issues only - I repeat that I am not maintaining it).

Best regards,

Petri

rgleason, 2020-01-19:
Petri,  I did what you said to do above and it built a file. Is there any problem asking Hakan if he can start a new crowdin sub project dashboard_tactics_pi using that POT file?
See http://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html#post3058433
II assume you are just currently using Dashboard_pi crowdin, but  how do you handle the Tactics_pi international definitions now?

Thanks.

canne, 2020-01-19:
Fine, good to hear that there is no technical issue over there, either. Regarding to have a Crowdin _dashboard_tactics_pi_ project I have nothing against that, I would be pleased. But please read the manifesto: I am working on version 2.0 and will not have resources to do back-port work on v1.5.11 or its patches anymore. So one may just wait for the next version before bugging the community.

If you consider that internationalized version, patch v1.5.111 is needed, pull-request **with test report** (I will not test) **in the** `pr151` **branch** will do the job. If it does not break the builds and meet the quality requirements, I'll tag it and out it goes.

Regarding _tactics_pi_ po-usage: I am not using them. I've just copied the C++ code, so I presume that they contain about 95% same strings than in Crowdin (which is behind). But not in the same positions: **warning** I am going to refactor all Tactics_pi code in the version 2.0. The modules are just too huge now, they need to be split in many small pieces for better maintenance., This means that if you do the work now for v1.5.111, you need to follow the refactoring caused changes in the v2.0. All that work would be much appreciated since I will not have time for it.

canne, 2020-04-01:
PR https://github.com/canne/dashboard_tactics_pi/pull/108 release https://github.com/canne/dashboard_tactics_pi/releases/tag/v1.5.111

canne, 2020-04-01:
Backport into dev200 https://github.com/canne/dashboard_tactics_pi/commit/8d5cf6ad4d06b564fb93394bd1e11c47eb1b3ac9

closed 2020-04-01 by canne