canne/dashboard_tactics_pi #105: cmake doesn't find wxWidgets_LIBRARIES

free-x, 2020-03-10:
Hi,

i'm trying to compile master on Debian 10. And CMAKE doesn't find wxWidgets

```
$ cmake ../
-- *** Building dashboard_tactics_pi ***
-- Build type: 
-- *** Will install to /usr/local  ***
-- *** Build Architecture is amd64
-- *** Staging to build dashboard_tactics_pi ***
-- Setting C++11 standard via cmake standard mecahnism
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  cmake/PluginConfigure.cmake:199 (FIND_PACKAGE)
  CMakeLists.txt:34 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so
--     Include: /usr/include
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.13/Modules/FindwxWidgets.cmake:963 (find_package_handle_standard_args)
  cmake/PluginConfigure.cmake:244 (FIND_PACKAGE)
  CMakeLists.txt:34 (INCLUDE)

-- Configuring incomplete, errors occurred!
````

Headers and Libraries 
```
$ dpkg --get-selections | grep wx
libwxbase3.0-0v5:amd64				install
libwxbase3.0-dev:amd64				install
libwxgtk-webview3.0-gtk3-0v5:amd64		install
libwxgtk-webview3.0-gtk3-dev:amd64		install
libwxgtk3.0-0v5:amd64				install
libwxgtk3.0-dev:amd64				install
libwxgtk3.0-gtk3-0v5:amd64			install
libwxgtk3.0-gtk3-dev:amd64			install
libwxsvg-dev:amd64				install
libwxsvg3:amd64					install
wx-common					install
wx3.0-headers					install
wx3.0-i18n					install
```

Regards
free-x

transmitterdan, 2020-03-10:
Same result on Raspberry Pi using Buster distro.

This may have something to do with gtk3 vs. gtk2

canne, 2020-03-10:
Hi, looks odd, indeed. 

But first, you'll find some packages for your respective systems from https://github.com/canne/dashboard_tactics_pi/releases/latest , including Raspberry Pi 4 Buster (eg. Debian 10 on __armhf__) - I don't have any feedback and if you can give some on those that would be appreciated!

Back to your problem: this is how I do it on Raspberry (right now, - I'm in a middle of a non-stable branch, please ignore the branch):
```
pi@pi:~/dev/dashboard_tactics_pi $ git status
On branch dev200chart1
Your branch is up to date with 'origin/dev200chart1'.

pi@pi:~/dev/dashboard_tactics_pi $ dpkg --get-selections | grep wx
libwxbase3.0-0v5:armhf				install
libwxbase3.0-dev:armhf				install
libwxgtk3.0-0v5:armhf				install
wx-common					install
wx3.0-doc					install
wx3.0-examples					install
wx3.0-headers					install
wx3.0-i18n					install

pi@pi:~/dev/dashboard_tactics_pi $ rm -rf build
pi@pi:~/dev/dashboard_tactics_pi $ mkdir build
pi@pi:~/dev/dashboard_tactics_pi $ cd build
pi@pi:~/dev/dashboard_tactics_pi/build $ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- *** Building dashboard_tactics_pi ***
-- Build type: 
-- *** Will install to /usr/local  ***
-- *** Build Architecture is armhf
-- *** Staging to build dashboard_tactics_pi ***
-- Setting C++11 standard via cmake standard mecahnism
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  cmake/PluginConfigure.cmake:199 (FIND_PACKAGE)
  CMakeLists.txt:34 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/arm-linux-gnueabihf/libOpenGL.so   
-- Found OpenGL...
--     Lib: /usr/lib/arm-linux-gnueabihf/libGL.so/usr/lib/arm-linux-gnueabihf/libGLU.so
--     Include: /usr/include
-- Found wxWidgets: -L/usr/local/lib;-pthread;;;-lwx_baseu-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_webview-3.0;-lwx_gtk2u_gl-3.0 (found version "3.0.4") 
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
-- Found BZip2: /usr/lib/arm-linux-gnueabihf/libbz2.so (found version "1.0.6") 
-- Looking for BZ2_bzCompressInit
-- Looking for BZ2_bzCompressInit - found
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.11") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/dev/dashboard_tactics_pi/build

make
...

[ 98%] Building CXX object CMakeFiles/dashboard_tactics_pi.dir/src/nmea0183/xdr.cpp.o
[100%] Linking CXX shared library libdashboard_tactics_pi.so
[100%] Built target dashboard_tactics_pi

```
In the continuous integration, from last night Ubuntu 18.04 platform on Travis (another branch), see this console log: https://travis-ci.org/canne/dashboard_tactics_pi/jobs/659920441?utm_medium=notification&utm_source=github_status

Hope this help, you can see that I keep it dead simple, out of the box and w/o GTK3! The best way to make sure that you can actually see something in you system is to compile from wxWidgets examples WebView based browser, as explained in the developer's guide (if your interest are the JS instruments - be ready to suffer... 😁 ): https://canne.github.io/dashboard_tactics_pi/docs/webview/developers/README.html#WebView-specific-on-Linux-__WXGTK__-&&-wxUSE_WEBVIEW_WEBKIT

Maybe not relevant, but I remember giving a hand on this same very subject on Ubuntu 16.04 LTS a few years ago and the person gave nice feedback of some unexpected dependencies back in the day: https://github.com/tom-r/tactics_pi/issues/19#issuecomment-357644991

If you find the magic key, please give feedback, thanks!

free-x, 2020-03-11:
@transmitterdan 
by the looks of it, you've done it. Would you care to explain your trick? ( https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601-4.html#post3092297)

@canne 
I have taken a closer look at your output and find that the libraries are found in /usr/local/lib. These are probably self-compiled artifacts

>Found wxWidgets: -L/usr/local/lib;-pthread;;;-lwx_baseu-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_webview-3.0;-lwx_gtk2u_gl-3.0 (found version "3.0.4") 

it does not even try to search in /usr/lib/[platform]*. Actually there are the libraries from debian packages

*/usr/lib/i386-linux-gnu/
*/usr/lib/x86_64-linux-gnu/

transmitterdan, 2020-03-11:
I did

git checkout v1.5.11

and then cmake was able to find wxWidgets. But I was building for Raspbian. I don’t know why the other branches do not work on Buster.

free-x, 2020-03-11:
indeed....1.5.11 can find
```
-- Found wxWidgets: -L/usr/lib/i386-linux-gnu;-pthread;;;-lwx_baseu-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_gl-3.0 (found version "3.0.4")
```
I have tried with "diff" to find a difference and Found
```
 IF(NOT DEFINED wxWidgets_USE_FILE)
-  SET(wxWidgets_USE_LIBS base core net xml html adv aui)
+    SET(wxWidgets_USE_LIBS base core net xml html adv aui webview)
 ENDIF(NOT DEFINED wxWidgets_USE_FILE)
```

If i delete webview from list then I can configure with cmake . But Compiling is erroneous 
Also I found some comments from @leamas where 
instead of wxWidgets_USE_LIBS wxwidgets_FIND_COMPONENTS is used Maybe this is the solution

leamas, 2020-03-11:
That was a bug more complicated than so. IIRC there was old,  pre cmake 3.0 code around using wxWidgets_USE_LIBS to define the libraries to use. However, cmake after 3.0 is using `find_package(wxWidgets COMPONENTS core base …)`. So what we did was to rename wxWidgets_USE_LIBS => wxWidgets_FIND_COMPONENTS} and changed the find_package invocation to  `find_package(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS})`

leamas, 2020-03-11:
Adding to above: the bug above was invisible in earlier cmake 3.x because there was an undocumented support for   wxWidgets_USE_LIBS. At some point this was dropped and the bug bite us.

canne, 2020-03-11:
@free-x, I am a bit confused now: you said that your are building in `master`, which (still) remain v1.5.11.

It has, indeed, in [its PluginConfigure.cmake#L148](https://github.com/canne/dashboard_tactics_pi/blob/c7656142b49c6bb38b3d1c681db9b2426f22193a/cmake/PluginConfigure.cmake#L148) `   SET(wxWidgets_USE_LIBS base core net xml html adv aui)`

In the current _default_ branch `dev200` (alpha stage) we have in same line [PluginConfigure.cmake#L148](https://github.com/canne/dashboard_tactics_pi/blob/b62ab4d8a88d197b451e710133c20d1b34c3fb9f/cmake/PluginConfigure.cmake#L148) `    SET(wxWidgets_USE_LIBS base core net xml html adv aui webview)`

where `webview` is required as back-end for JavaScript based instruments. I presume that **this** is what you want to do: test the alpha-stage v2.0 (which does not compile for you) and not to compile v1.5.11 (which does not have issues, finally, right @transmitterdan ?).

@leamas is absolutely right and `wxWidgets_FIND_COMPONENTS` is, according to _cmake 3.x_ documentation the right way to go.

I tried to move to use it in December. Alas, I could not make it working in my CI environments (travis and appveyor). Also it did not work on my Windows 10 with locally compiled wxWidgets with cmake v3.13.5 (on Raspberry v3.13.4, FYI, from the above build: 
[CMakeCache.txt](https://github.com/canne/dashboard_tactics_pi/files/4320653/CMakeCache.txt) ).

I rolled back, and on four different build environments I am using, `wxWidgets_USE_LIBS` works fine so I would not be personally too eager to fix something which works - before it breaks! But please go ahead and try `wxWidgets_FIND_COMPONENTS` on Debian and give feedback here, that would be an appreciated knowledge if you can make it work.

Meanwhile, you can take a look at this configuration in my Ubuntu 18.04 LTS build platform on Travis: [travis.yml#L16](https://github.com/canne/dashboard_tactics_pi/blob/b62ab4d8a88d197b451e710133c20d1b34c3fb9f/.travis.yml#L16) and doublecheck your libraries against it - I can remember spending many days to get it right: there is some weird dependency in WebView, probably caused by the fact that they are using WebKit1 and not WebKit2 on Linux - even though they have the code for WebKit2 incorporated!

>`off-topic` - I just wonder is anybody actually still working on WebView _chez_ wxWidgets? Anyway, it is highly limiting factor for JS development, nothing modern works on it cross-platform and polyfill is an everyday blues for Windows back-end: IE! 😒

canne, 2020-03-11:
@xfree-x, you are right about the architecture links missing on my RPI: I fired it up and I found that I have actually on it a `wxWidgets-3.0.4/gtk-build` : 
[config.log](https://github.com/canne/dashboard_tactics_pi/files/4320806/config.log)

Seeing that @transmitterdan was able to build the v1.5.11 on RPI Buster with the default libraries and it builds for you as well, we can perhaps conclude that for some reason the `libwxgtk-webview3.0-gtk3-dev:<architecture>` does not fly on Debian 10 based distribution and that's why I was obliged build it from scratch (sincerely, I can't remember was this the reason I started to build this on a RPI!) and, hop, cmake even with `wxWidgets_USE_LIBS` works again? On Ubuntu 18.04 LTS, somehow it works out of the box.

Interesting...

leamas, 2020-03-11:
We do have a common understanding here, right? That  it's not only about renaming wxWidgets_USE_LIBS => wxWidgets_FIND_COMPONENTS. 
butl also to change to  `find_package(wxWidgets COMPONENT${wxWidgets_FIND_COMPONENTS})`?

leamas, 2020-03-11:
Another thing which might bite is the update-alternatives  setting for wx-config.

canne, 2020-03-11:
@leamas  - absolutely, fully agree, we're on the same page! That's exactly what I tried in December but failed. Maybe I did something silly. Since my main objective in `dev200` (in alpha2!) branch is JS (actually TypeScript) development, I will not be able to spend time on cmake testing, it is really time consuming business! Let's hope that this thread will bring up more light like your suggestion (and hopefully cross-platform - important for me - .cmake mods).

canne, 2020-03-11:
Related to the above : eventual PRs (if any) to `pr200` branch please to keep them safe - I have multiple flying fishes above `dev200` right now and your PR would be just probably lost on `dev200` which is for JS work. Thank you for your comprehension.

leamas, 2020-03-11:
Sorry, no  time to make a PR. You might be interested in the ci script for oesenc_pi; it builds on bionics which basically is Debian 10. The script:  is at https://github.com/bdbcat/oesenc_pi/blob/master/ci/circleci-build-debian.sh, the  control file used is https://github.com/bdbcat/oesenc_pi/blob/master/ci/control

canne, 2020-03-11:
Thanks, I'll keep that in mind, the usage of update-alternatives is also something to be done,of course, in this case. Will be done sometime this summer once I get back to C++...

leamas, 2020-03-11:
There is this dream about the Nordic Summer...

canne, 2020-03-12:
Ok, let's make it Nordic Winter then, more realistic... Meanwhile, submissions to `pr200` are welcome, even if not cross-platform tested.

canne, 2020-04-23:
Memo for future work on Debian: https://github.com/OpenCPN/OpenCPN/blob/9be7efe9641f516f8efabb6401445970595474a6/ci/generic-build-debian.sh

canne, 2020-05-09:
I am studying this issue through Ubuntu 18.04LTS, 20.04LTS and Raspberry Raspian Buster, all derived from Debian - sorry that I do not have the courage and time to start installing that now and really willing to leave this to this summer 😁 ):

https://github.com/canne/dashboard_tactics_pi/commit/d06f3d11275db93cb17b689b92f21ca175faecd5 implements the method already discussed [above](https://github.com/canne/dashboard_tactics_pi/issues/105#issuecomment-597881372).

It resolves the issue on Ubuntu 20.04LTS.

The resolving link in the produced .so-file is

```
libwx_gtk3u_webview-3.0.so.0 => /lib/x86_64-linux-gnu/libwx_gtk3u_webview-3.0.so.0
```

There is no ambiguity in Ubuntu 20.04LTS which is GTK3-only installation, so is OpenCPN in it.

But on Ubuntu 18.04LTS (and - I presume - in Debian 10) the issue is with the two alternative versions of wxWidgets libraries, one set for GTK2 and the other for GTK3. One can select between the two with `sudo  update-alternatives --config wx-config`. If one selects GTK3, _DashT_ v2.0.0alpha builds. If one selects GTK2, only _DashT_ v1.5.11 builds on this platform:

Helas, only `libwx_gtk3u_webview-3.0.so.0` is present in the system and available for installation. There is no equivalent for GTK2. 

But it is precisely GTK2-libraries which are needed in OpenCPN v5.0 and v5.1 on Ubuntu 18.04 LTS and on Raspberry Raspian Buster. It is not possible to have both GTK2 and GTK3 in the same binary. Before loading, plugin manager of OpenCPN is going through the ELF (equivalent of `readelf`-command) and if wrong versions are found (also for wxWidgets) the plug-in is deemed "not compatible".

Therefore, one is obliged, on Ubuntu 18.04 LTS to declare GTK2 usage. If one builds v1.5.11 of _DashT_ it works. If one builds v2.0.0alpha which requires `webview` for wxWidgets, the build fails.

**Conclusion** - for FAI (fully automated installation) _DashT_ v2.0 will be available only if wxWidgets WebView run-time library and development library are available by the system installation tools in the same GTK version OpenCPN is using.

For others, like probably Debian 10 mentioned in the title, the remedy is manual and requires either to compile and build wxWidgets on the target (like I did on myRPI 4B), or to find a [back-ported library for the distribution](https://backports.debian.org/Instructions/).

closed 2020-05-09 by canne

canne, 2020-05-10:
Note for future issues of this kind (this is quite confusing, indeed...): I have collected the above wisdom into a Linux [shell script](https://github.com/canne/dashboard_tactics_pi/blob/6904e1b234aec2a48f61005f271ec64fc3f31b31/data/instrujs/scripts/linux/o-platforminfo.sh) which makes most of the above analysis (minus the _alternatives_). The below file is the output report from a Raspian Buster w/ OpenCPN v5.1 beta.
[report.txt.gz](https://github.com/canne/dashboard_tactics_pi/files/4606613/report.txt.gz)

canne, 2020-11-15:
Link reminder my comment Ubuntu 18.04LTS : https://github.com/seandepagnier/rtlsdr_pi/issues/25#issuecomment-624861477

canne, 2020-11-15:
Reopening: I started to study the port of OpenCPN v5.2.4 on Raspberry OS 64-bit beta on RPI 4 w/ kernel `Linux raspberrypi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux` - with objective to port also dashboard_tactics_pi on it. But first OpenCPN should work and the issue in OpenCPN build is exactly the same as explained above  - a major headache, all these GTK2 / GTK3 based systems!

canne, 2020-11-15:
References from OpenCPN:

https://github.com/OpenCPN/OpenCPN/issues/1898 and patch to fix it https://github.com/leamas/OpenCPN/commit/0b14869320e7b54ea979d8c8090471518373e612

canne, 2020-11-21:
Reopening is **not necessary** but the conclusions above remains for the the port, to be used with an OpenCPN v5.2.4 on Raspberry OS 64-bit beta on RPI 4 w/ kernel `5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux ` 

- Pull request #144  was first implemented (thanks!);
- The  `update-alternatives` method I explained in https://github.com/seandepagnier/rtlsdr_pi/issues/25#issuecomment-624861477 remains applicable - this is yet another crazy mixed GTK2 / GTK3 system (just wonder why they torture us with these _alternatives_?).

Build of packages is successful. But they need to be tested in another arm64/aarch64 system since the OpenCPN v5.2.4 builds by default with GTK2 if that this available. And here it is. Yes, yes, one can force it with `-DOCPN_FORCE_GTK3:BOOL=ON` but that's not what the user is presumably getting from the repository. Anyway, the OpenCPN part on RasbperryOS64 is out of scope of this ticket and the plug-in, and we move to the Ubuntu 20.04LTS Focal 64-bit platform on RaspberryPi 4 to test the same:

The test pass, no need to any GTK2/GTK3 tricks on Ubuntu 20.04LTS Focal  and no need thus to reopen this ticket.

leamas, 2020-11-27:
While working, the update-alternatives isn't really ideal. A major drawback is that when doing a local build it affects the state of the wx-config alternatives link.

There three better mechanisms which can be used:

  - The WX_CONFIG environment variable could be set to the full path of the wx-config utility to use, for example _/usr/lib/x86_64-linux-gnu/wx/config/gtk3-unicode-3.0_ for a gtk3 build
  - The wxWidgets_CONFIG_EXECUTABLE cmake command line variable could be set the wx-config path like for WX_CONFIG.
  - The wxWidgets_CONFIG_OPTIONS command line variable could be set to include the string *--toolkit=gtk3*  (or ...=gtk2)