canne/dashboard_tactics_pi #110: %Polar not right on NKE TL25

JoostFrouFrou, 2020-04-08:
using Open CPN 5.0.0 + 9065270 and DashT v1.5.11

%Polar on TL25  is always 0 (zero) when sailing upwind example $PNKEP,03,44.0,112.9,0.0*56

when sailing downwind $PNKEP,03,140.0,0.0,190.8*68 results in %polar of 204

looking closer it happens that Target VMG results in 191 so it looks like there is something mixed up.

[testpolar.LOG](https://github.com/canne/dashboard_tactics_pi/files/4450546/testpolar.LOG)

canne, 2020-04-08:
Thanks @JoostFrouFrou for having taking the time to report this issue. It will be fixed (I have a TL25) but I cannot tell when; I cam currently working on something else while [waiting for the OpenCPN v5.1](https://github.com/canne/dashboard_tactics_pi/blob/dev200/docs/developers/README.md) which I would use for a motivation to revamp the code base (it is is still carrying the v5.0.0 Dashboard in it and thus suffering from a software bloat). Meanwhile, please use Tactics v1.11 if you need this function.

JoostFrouFrou, 2020-04-09:
Hi Canne, my latest version Tactics (1.0008) has the same problem. It looks like NKE uses a different nmea notation. 
downwind: $PNKEP,03,141,0,97*49 results in 97% Polar performance
upwind :$PNKEP,03,44,0,71*75 results in 71% Polar performance
as you can see NKE projects the valid Polar% is always at the end of the sentence.

I hope this is valuable info for you

canne, 2020-04-09:
Hi Joost, thank you for this feedbac. It is indeed very useful. It looks like this is an issue with tactics_pi.

FYI, in the repo there is still the patch level 11 (not 8) from which I have derived these functions into DashT v1.5.11. That (probably) means that the issue would be the same in tactics_pi v1.0.11 still. Let's see: @tom-r who knows about NKE (I have only my TL25, everything else is Raymarine) says [in his document](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/tactics_pi.pdf) section 3.7: 
[image]
You can see that a) there is a different field for upwind and for downwind performance and b) they are supposed to use decimal dot notation.

I cannot sail now for the reasons we all know but if I look my records with tactics_pi from Cagliari 2018 I can see `$PNKEP,03,50.0,12.6,0.0*6D` type of sentences on the other hand, and `$PNKEP,03,50.0,0.0,19.6*66` on the other hand. It was one, long leg _près serré bâbord_, so I am confused why I have the two fields? And the percentage of performance? Come on, I was sailing like a champion! But I only show the optimal angle, 50 degrees to the helmsman (me) on TL25 so this terrible insult went unnoticed...:grin: 

Long story short, I have, of course a fork of tactics_pi: even if I do not want to work on DashT while awaiting 😴  for OpenCPN v5.1 I will take a look in Thomas' code base if I can see an issue there and maybe contribute back to it. Or maybe he will share his NKE wisdom with us.

JoostFrouFrou, 2020-04-09:
Hi Canne, the 2 nmea sentences from me are logged from NKE, they differ in Target upwind angel 44 and target downwind angel 140. As you can see is the polar performance for both in the last section. This works on the TL25. If Tactics projects as we expect in the forlast section for upwind performance, the last section becomes 0 which is displayed on the TL25.

canne, 2020-04-09:
Dear Joost, that differs a bit from what Thomas has document (and probably implemented). Ok, I will take a look in his code and eventually I can test it with my TL25 once I get an access to it.

canne, 2020-04-09:
For the record only, a couple of observations.

First, the few available sources ([1](http://nke-marine-electronics.fr/wp-content/uploads/2017/01/gyrpilot-graphic-UK.pdf)) ([2](https://www.hisse-et-oh.com/sailing/canaux-performances-de-maxsea-sur-tl-25-nke)) confirms the above referenced Thomas' document declaring `$PNKEP,03` sentence's fields correctly.

Second, once that sentence is extracted from the above log from Joost, it does make sense logically with the above definitions but it is the values, which are jumping in unpredictable way. For example, downwind we can observe values like `627.5` ! 
[polar_nke_tl25_joost-pnkep03only.txt](https://github.com/canne/dashboard_tactics_pi/files/4457680/polar_nke_tl25_joost-pnkep03only.txt)

Let's compare that with the extracted `$PNKEP,03` sentences from my gig in Cagliari 2018, remember that this is one single, long leg with stable wind, the values of the fields Joost is looking for are jumping like mad (without me noticing). I reckon we use the here same version of tactics_pi seeing the data of this recording: 
[cagliari_2018_babord_leg-pnkep03only.txt](https://github.com/canne/dashboard_tactics_pi/files/4457735/cagliari_2018_babord_leg-pnkep03only.txt)

```
...
$PNKEP,03,50.0,15.8,0.0*64
$PNKEP,03,50.0,0.0,19.1*61
$PNKEP,03,50.0,0.0,27.6*6B
$PNKEP,03,50.0,8.8,0.0*58
...
```
Definitely there is something wrong here. Let's try to find out what.

JoostFrouFrou, 2020-04-09:
I did not mention that I did not sail but did my run on the engine so when I motor upwind I get tremendous performance as DonaldTrump should say

canne, 2020-04-09:
😁

JoostFrouFrou, 2020-04-09:
If I scroll through the cagliari files I see most of the PNKEP,3 data is upwind because it mention the Target angel 50 which should be upwind, sometimes there is a 157 which is the Target downwind angle which is strange if you sail upwind......

JoostFrouFrou, 2020-04-09:
the only thing I can think of is that the windsensor or the compas has a hicup which leads to a miss calculation which leads to a wrong PNKEP,03

Do you have the complete log with the other nmea sentenses

canne, 2020-04-09:
True. It is clear now: the algorithm used in numerical (percentage + theoretical polar speed) instrument which I am looking at is using [this algorithm](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/performance.cpp#L214) (pointing now to original upstream sources, not my source), while the values for the PNKEP03 are generated [here, with a different algorithm](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/tactics_pi.cpp#L5153) which has some fixed components (60 degrees upwind, 120 degrees downwind).

I need to map those to the complicated vector graphics we like to think with (from the doc):
[image]

Here's my complete vdr recording, zipped: 
[cagliari_2018_babord_leg.zip](https://github.com/canne/dashboard_tactics_pi/files/4457942/cagliari_2018_babord_leg.zip)

Read also what the author says [in this line of the original code](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/tactics_pi.cpp#L5206) - the statement is quite clear about the maturity of PNKEP sentences.

Can you feed your TL25 from a NKE regatta computer (which I do not have, do you?). What does it say, what value from the above diagram it is referring to?

JoostFrouFrou, 2020-04-09:
if I run the file through de VDR player and filter for example on MWV it fluctuates 10 or 20 degrees at a time, that is not possible in real life

JoostFrouFrou, 2020-04-09:
Oh, I do not have a regatta processor

JoostFrouFrou, 2020-04-09:
next thing which catch my eye, the track line is not smooth but looks like a saw.....

JoostFrouFrou, 2020-04-09:
I see AIVDO sentence passing by, this is the AIS sentence from own boat, it has GPS info from a second GPS which normaly leads to problems, it is recomended to put this in the reject filter

JoostFrouFrou, 2020-04-09:
I see heel going from port to startboard which is strange with the wind coming from port..... something is not right

canne, 2020-04-09:
No problem, from the vector graph and the algorithms it is clear now: we can see that the PNKEP03 algorithm makes perfect sense, at least from its attempt to show the percentage: Since it is showing the target VMG angle, it is projecting also the speed to this vector and shows the percentage of it. I guess that the idea is to show how far you are from ideal.
However, that ideal seems to be fixed angle while the polar file for a given speed may think otherwise. I am bit puzzled with the fixed angles I see in the code for now.

>Above I referred to the boat speed instrument, where I should have referred to the [POLARTARGETVMG algorithm](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/performance.cpp#L234):

[image]

What comes to the parameters and my VDR log file; yes, at that moment I had three GPS devices running on-board. And many things happen on Golfo Cagliari... But I think that is not an issue. if you check the algorithm, it takes only TWA, TWS, StW and the polar file. Of course, the TWA should be stable and reliable. But on the other hand, it is projected to the polar graph where one has no big jumps because of that.

Now, we are getting closer to the requirement capture: do you think that your display should say exactly the same as above POLARTARGETVMG numerical instrument? And what does it say for you? Is it stable? Does it show what you think it should be?

canne, 2020-04-09:
On numerical instrument target VMG  from polar is calculated using TWA and TWS.

To build PNKEP03 sentence, tactics_pi calculates VMG from polar using fixed apparent wind - 60 degree upwind and 120 degrees downwind - and TWS.

JoostFrouFrou, 2020-04-09:
to be honnest, I never checked if the polar% corresponds with the table.....

canne, 2020-04-09:
No worries, I can run your data stripped from PNKEP sentences against my own boat's polar and play with the algorithm. to see if the data on POLARTARGETVMG is the same than the generated PNKEP03 senteces. I would say that would be the requirement, for me, at least. I keep you posted.

canne, 2020-04-09:
Please try this build 
[tactics_pi-1.0.11001-ov50-win32.zip](https://github.com/canne/dashboard_tactics_pi/files/4458775/tactics_pi-1.0.11001-ov50-win32.zip)

forked from tactics_pi v1.0.11 (Thomas' repository) + my modifications.

(make sure that you have your original installer or make backups of .dll files of your orignal tactics_pi)

One gets the same result on the POLARTARGETVMG instrument and on the generated PNKEP,03 sentences.

Test input is your vdr file _without_ previously generated PNKEP-sentences: 
[polar_nke_tl25_joost-nopnk.zip](https://github.com/canne/dashboard_tactics_pi/files/4458793/polar_nke_tl25_joost-nopnk.zip)

Gives like
```
$PNKEP,03,45.0,87.6,0.0*65
$PNKEP,03,45.0,87.2,0.0*61
$PNKEP,03,45.0,87.6,0.0*65
```
with my boat's polar file.

[image]

tom-r, 2020-04-10:
Thanks Canne for looking into it.
As you already found out, I took the PNKEP sentence description from a NKE documentation.
I was never able to compare it to real data generated by a NKE "instrument" (processor) ...
And I don't use this one on my own boat.
Concerning the fixed angles 60°upwind, 120° downwind :
There was a reason for this, I don't remember right now, I have to recap it.
I **_think_** it was the definition of "VMG" = best possible performance up-/downwind, _independent on the angle_. So I needed only 2 clear values for upwind/downwind.
Something like that ...
I can look into it next monday...

BR,

Thomas

PS: I was supposed to prepare and haul my boat this week, but I can't even go there, borders to Italy are closed ...

canne, 2020-04-10:
Hi Thomas,

Same thing here, no access to boat,  coastline is closed. Anyway, NA flags are hoisted since three weeks and whales have returned to _calanques_ (something good from this)! Good to have something loosely related to sailing back home! 😄 

I was thinking, like you that the fixed angles must have a meaning. Maybe one would then need to look for these optimal values from the polar instead of hard coding them as theoretical ones?

The version I made for Joost is pulled from your repo as upstream/remote and I applied this https://github.com/canne/tactics_pi/commit/0ad5f1a6424a895a275b59b009f20ae6bdf79fc7 to make the output of the NKE sentence being the same as the numerical instrument.

I'm not making a PR but I let you think about it - in NKE dependencies you are the reference of course; there is no information available how their thinking is going!

Best regards,
Petri

JoostFrouFrou, 2020-04-10:
Hi Petri,

It looks good on the TARGETVMG instrument, PNKEP,03 is displayed over there for up and downwind.

But now you use the PNKEP,03 for TARGETVMG instead of output to display POLAR% on the TL25, which has still the same problem, downwind gives a value, upwind gives no value.

You showed me some changes/details on program territory which is fortunately not my thing.

Joost

canne, 2020-04-10:
Thanks Joost, for testing,

As I said, there has been no change in TARGETVMG numerical instrument. The modification consist solely to send the very same TARGETVMG percentage and the optimum angle of it in the PNKEP,03 sentence.

In other words, what you see in TARGETVMG instrument, is now going out in PNKEP,03 sentence.

Currently the issue is to define what exactly is the requirement. You seem to say that some POLAR% should be sent instead of TARGETVMG%. I am confused, percentage of what, speed, angle or VMG? We should stick to the terms of Thomas' manual, otherwise this is going to be difficult.

OK, let's wait; Thomas will hopefully remember why fixed 60/120 AWA angles were used in the first place.

Once the requirement has become clear and preferably explained using the very same terms of the above, excellent vector diagram extracted from Thomas' manual I am ready to make a new effort for its implementation now when I have the tactics_pi code repository checked out.

Best regards,
Petri

canne, 2020-04-10:
For the record, before I move to work on something else: I tested against the [above bug report](https://github.com/canne/dashboard_tactics_pi/issues/110#issuecomment-611944579) that while downwind target VMG percentage value appears in PNKEP,03 sentences the upwind target VMG percentage value does not.

With the [above test bed](https://github.com/canne/dashboard_tactics_pi/issues/110#issuecomment-611769962) (thus with my own polar file and the same test file stripped from PNKEP,03 senteces) ;

**Test procedure**: play the test file with VDR, leaving NMEA debug window visible, the area of test is the turning point from downwind leg to upwind leg ;
<hr />

 1. Run the tactics_pi using OpenCPN as debugger executable for VS ;

 2. Place first breakpoints on patch tactics_pi.cpp lines [L5173](https://github.com/canne/tactics_pi/blob/0ad5f1a6424a895a275b59b009f20ae6bdf79fc7/src/tactics_pi.cpp#L5173) and [L5176](https://github.com/canne/tactics_pi/blob/0ad5f1a6424a895a275b59b009f20ae6bdf79fc7/src/tactics_pi.cpp#L5176) ;

 3. Before the turn (downwind) second breakpoint shall get hit ;

 4. After the turn (upwind) first breakpoint shall get hit ;

**Result** : PASS.
<hr />

Record and observe by filtering the generated PNKEP-sentences with the OpenCPN's NMEA Debug Window. Record both the text output from the debug window and take the screenshot on both sides of the downwind / upwind turning point ;

[2020-04-10_nmeadbgwin_copy_downwind_upwindturn.txt](https://github.com/canne/dashboard_tactics_pi/files/4462763/2020-04-10_nmeadbgwin_copy_downwind_upwindturn.txt)

[image]

[image]

Pass criteria:

1. downwind leg should procedure sentences such as `$PNKEP,03,35.0,95.7,0.0*60`
2. upwind leg should produce sentences such as `$PNKEP,03,30.0,98.4,0.0*6B`
3. both generated sentences pass the checksum test in https://nmeachecksum.eqth.ne

**Result**:  PASS.
<hr />

**Conclusion**: the modified algorithm works as expected, generating and sending out the `mPercentTargetVMGdownwind` and `mPercentTargetVMGupwind` consistent with tactics_pi user's manual sections 3.5.4 and 3.7.

It could be useful to repeat the above test with other polar files than mine which is honed out during several years now and does not contain discontinuities. It is not necessary to repeat the VS breakpoint test, it has no dependencies on the polar file, only TWA.

JoostFrouFrou, 2020-04-11:
Hi Petri,

I am afraid that the VMG and Polar Performance get mixed up now.

As I know it with the Polar Performance % I can directly tell something of the sailtrim, important on deck of the sailboat. VMG% is the %vector towards the wind and is important for the man down below responsible for the tactics. As I see it I can sail with 100% Polar Performance and still have 70% VMG. 

I have send an email to NKE service in France (Joel Bars) with the question of the PNKEP,03 sentence, why they use the last data spot (data3 in OpenCPN) for both, upwind AND downwind Performance instead of data2 for upwind Performance and data3 for downwind Performance (as it is declared in the manual of Tactics).

I do not have a reaction from him yet.

Joost

canne, 2020-04-11:
Thanks, Joost,

that's a good initiative to get feedback of NKE to confirm what are the actual definition of the PNKEP,03 sentence contents for them! 👍  Please report their answer here if you get any.

I fully understand your needs and that's why I have never tried to get the other PNKEP,03 values on my TL25 because they do not interest me (albeit I can understand their meaning). Only the optimal angle and the Polar **Speed** percentage (section 3.5.1 in Thomas's manual) are used in my sailing:

The optimal angle (section 3.5.3) I get already from TL25 (via PNKEP,03) and that's enough for the helmsman; the polar speed percentage I follow from the OpenCPN screen if I have time (sections 3.5.1. and 3.5.8) for the sail trim. Being a solo sailor, I have found that it is more productive to watch my jib's edge than the computer screen, though...

I am sorry that you feel disappointed, but I can assure you that I have not changed or mixed anything in the original code of _tactics_pi_ what comes to these "percentages" in the patch of it I sent to you above: I have just implemented what is defined in the original user's manual,in sections 3.5.4 and 3.7. - you should rather challenge my test above against those definitions.

If the value you would like to see is different than what is sent out does not mean that the code is not working as defined in the manual.

The user's manual is highly recommended reading regarding the meaning of each polar percentage. and their usage. You can get the original document of Thomas from his repository, [this is the link to the pdf-file](https://github.com/tom-r/tactics_pi/blob/master/tactics_pi.pdf).

Now, after reading Thomas' document sections 3.5 and 3.7 you do not agree with his definitions - or you are not happy with the fact that you do not get polar speed percentage on your TL25, that's a discussion which should be held in Cruisers Forum - you will get the community's regarded opinions on this matter.

GitHub is the place where the programmer - me, for example - implements what has been specified when the requirements for what needs to be implemented are clear. For example, if NKE says something else than what Thomas has found out for PNKEP,03 sentence, he will certainly update his document.

After which I can implement my code and make a pull request, in order to help him in testing since I have now the source code still on my machine.

I will follow this thread (but not threads on CF, unfortunately since I need to work on something else now).

JoostFrouFrou, 2020-04-11:
Hi Petri,

I do not argue the definitions, my problem is that I cant get it on my TL25, only downwind is displayed, upwind is always zero on the TL25.

I think France and NKE are in lockdown so I will wait for their answer and let you know.

Maybe I will ask on CF if someone else has the same problem as me with Polar% and TL25 to confirm the problem.

Kind regards,

Joost

canne, 2020-04-12:
For the record (only) : I managed to get from my boat my own TL25's User's Guide "_Multifunction TL25 and its remote control - Product reference 90-60-244 white / 90-6-256 black_".

It defines, in section _1.13.4 Frames identified by the TL25_:

| NMEA Frame | Description | TOPLINE channels created |
| -- | -- | -- |
| $PNKEP,01 | Target speed | _VIT_CIBLE._ |
| $PNKEP,02 | Heading at next board | _CAP_AUTRE_BORD._ |
| $PNKEP,03 | Optimum upwind angle | _ANGLE_OPT_VENT, REND_PRES, REND_POLAIRE._ |
$PNKEP,04 | Angles to optimise the CMG and VMG and gain | _ANGLE_OPT_CMG, ANGLE_OPT_VMG,GAIN_ROUTE_CMG, GAIN_ROUTE_VMG._ |
| $PNKEP,05 | Current direction and speed | _DIREC_COURANT, VITES_COURANT._ |

Not very helpful, does not say anything about the input fields, only about the TOPLINE channels - the pair of _REND_PRES_ and _REND_POLAIRE_ does not make any sense, where is a _REND_PORT_ (portant = downwind ; près = upwind ; rend = rendement = efficiency, yield) or equivalent ❔ 

Maybe it makes sense to somebody who has a possibility to see the "_TOPLINE_" channels in their NKE installation. Please execute the [above test procedure with the test data](https://github.com/canne/dashboard_tactics_pi/issues/110#issuecomment-612155500) and tell what "_TOPLINE_" tells you once your NKE installation gets tactics_pi generated PNKEP,XX-sentences.

BTW, In the [earlier reference above for the gyropilot](http://nke-marine-electronics.fr/wp-content/uploads/2017/01/gyrpilot-graphic-UK.pdf) the table is, actually exactly the same than in my document.

In this [Signal K discussion thread post from 2018](https://groups.google.com/d/msg/SignalK/balc8Fw4VOo/pY_3Eh4PAgAJ) they have ended up to copy the fields of PNKEP sentences from Thomas' documentation... However, in the current state of testing of the conversion plug-in they have  get stuck to [test only the target angle field](https://github.com/SignalK/signalk-parser-nmea0183/blob/71bade14785e4d4439e823390161d3e70ae63bed/test/PNKEP.js#L53), and the NMEA-0183 conversion plug-in has the [NKE Performance data grayed out](https://github.com/SignalK/signalk-parser-nmea0183).

So we are not alone. This mess is caused by quite counterproductive NKE's information hiding policy.

JoostFrouFrou, 2020-04-12:
[ACTUALITES_2014_NKE.pdf](https://github.com/canne/dashboard_tactics_pi/files/4466274/ACTUALITES_2014_NKE.pdf)

In French

canne, 2020-04-12:
Bien, cela me semble être clair maintenant: They join upwind and downwind under one variable in TOPLINE protocol, REND_PRE. With that French logic in the variable name...

Remains the mystery what NKE expects from the infamous PNKEP,03 sentence, then?

Maybe it should be, in this case:
```
$PNKEP,03,x.x,x.x,x.x*hh<CR><LF>
           |    |   \ polar speed performance TWA/TWS from 0 to 99%
           |    \ performance upwind or downwind from 0 to 99%
           \ opt. VMG angle 0 à 359°
```
instead of the actual
```
$PNKEP,03,x.x,x.x,x.x*hh<CR><LF>
           |    |   \ performance downwind from 0 to 99%
           |    \ performance upwind from 0 to 99%
           \ opt. VMG angle 0 à 359°
```
or maybe even
```
$PNKEP,03,x.x,x.x,x.x,x.x*hh<CR><LF>
           |   |   |    \ polar speed performance TWA/TWS from 0 to 99%
           |   |   \ performance downwind from 0 to 99%
           |    \ performance upwind from 0 to 99%
           \ opt. VMG angle 0 à 359°
```
❓ 

Mystery.

It would not be too complicated to make a special version of the tactics_pi to try this out.

But since the NA flags prevent me to sail, would you be able to test this hypothesis? It has chances to fail, of course.

JoostFrouFrou, 2020-04-12:
no problem, if you make a trail version, I will test it

Joost

canne, 2020-04-12:
Bingo! Here:  [SailGRIB WR page 91](http://doczz.fr/doc/285478/manuel-d-utilisation)
```
Voici un exemple d’affichage sur un Multigraphic :`
SailGrib WR V2.1.0
91 / 107
SailGrib WR
Navigation and Weather Routing for and by sailors
Le Gyropilot Graphic peut recevoir des données de performance dans un format statique.
Pour activer ce flux de données, allez dans les Paramètres NMEA puis Paramètres NKE et
cochez la case Gyropilot Graphic.
Les données suivantes sont envoyées vers le Gyropilot.
```
```
Phrase 1
$PNKEP,01,x.x,N,x.x,K*hh
           | Vitesse cible en Nœud
              | Vitesse cible en km/h
```
```
Phrase 3
$PNKEP,03,x.x,x.x,x.x*hh
           |Angle optimum de 0 à 359°
               | Rendement du VMG au près/portant en %
                   | Rendement polaire en %
```
```
Phrase 5
Direction et vitesse courant atlas
$PNKEP,05,x.x,x.x,N,x.x,K*hh
           |Direction courant de 0 à 359°
               | Vitesse courant en Noeud
                     | Vitesse courant en km/h
```

canne, 2020-04-12:
OK; here we go, a **test** version [tactics_pi-1.0.11002-ov50-win32.zip](https://github.com/canne/dashboard_tactics_pi/files/4466665/tactics_pi-1.0.11002-ov50-win32.zip)

Implementing above "Phrase 3": 
[Memo_pnkep03mods_2020-04-12.zip](https://github.com/canne/dashboard_tactics_pi/files/4466678/Memo_pnkep03mods_2020-04-12.zip)

**Note**: it creates tons of debug messages in `C:\ProgramData\opencpn\opencpn.log`.

Here's an example of the log file with my run of one single loop around:  
[opencpn-log_2020-04-12_182300_tactics_pi_11002.zip](https://github.com/canne/dashboard_tactics_pi/files/4466671/opencpn-log_2020-04-12_182300_tactics_pi_11002.zip)

It would be good to see your polar's version of the log file (as zipped file), no need to go out: a run even with your already recorded data would do - there are interesting points in it:  
[polar_nke_tl25_joost-nopnk.txt](https://github.com/canne/dashboard_tactics_pi/files/4466668/polar_nke_tl25_joost-nopnk.txt)

>It is **without** PNKEP messages because they are generated by tactics_pi - they should not be in the test recording already!)

Last but not least - does TL25 show any of this ❔ ❗ 

My test run before build with screen shots, for completeness sake:

**Downwind**:

[image]

**Against the wind** (!):

[image]

**Upwind**:

[image]

JoostFrouFrou, 2020-04-13:
Hi Petri,

I tested it at home, and it looks good, with upwind it gives Polar% on 'data3' 

I will test if the TL25 works later this day.

I installed the SailGrib App on my mobile to, interesting App I think, at least they give better insight in NKE nmea building.

I let you know as soon as possible

Joost

JoostFrouFrou, 2020-04-13:
Petri,

It works perfect, both upwind and downwind Polar% are projected on the TL25 as it should.

If you like a logfile of OPEN you have to tell me how I can make this.

Joost

canne, 2020-04-13:
Perfect!

Please attache here for the record the `opencpn.log` file ziped:

1. stop OpenCPN
2. copy `C:\ProgramData\opencpn\opencpn.log` on your desktop
3. right click on it -> Send To -> Compressed (zipped) folder
4. Drag and drop the resulting `opencpn.log.zip` here, in a new comment

I will make you a new version after having inspected that there are no issues reported in it. (The version above is filling the `opencpn.log` file rapidly and you should not let it run for a long time).

JoostFrouFrou, 2020-04-13:
Petri,

Running the file without PNKEP,03 you send me results in 'no polar data' in upwind situation because it falls outside the boundaries of the Polar Diagram which starts at 30 degrees TWA I think.

It was recorded upwind on motor ......

But if I play another route with PNKEP stripped it is good.

[opencpn.zip](https://github.com/canne/dashboard_tactics_pi/files/4469615/opencpn.zip)

Joost

canne, 2020-04-13:
Thanks, Joost,

Indeed, not a single "NaN" for calculated polar speed in this log file. Works perfectly with your polar file. You can now remove this test version `tactics_pi-1.0.11002-ov50-win32.zip` from your computer, it is not good for regular use because much too verbose.

For the record (you can ignore this) - I will not commit to my repository or create a PR for this version but here is the patch how one can get from 1.0.11001 (in repo) to v1.0.11002 with heavy debug logs: 
[0001-Aligning-polarvmg-target-instru-and-PKENP-03-output.zip](https://github.com/canne/dashboard_tactics_pi/files/4469715/0001-Aligning-polarvmg-target-instru-and-PKENP-03-output.zip)

I'll remove the debug printing, make a new version and report you back.

canne, 2020-04-13:
[tactics_pi-1.0.11003-ov50-win32.zip](https://github.com/canne/dashboard_tactics_pi/files/4469785/tactics_pi-1.0.11003-ov50-win32.zip) - same as before but without debug logs.

>This is **not** anything "_official_" or coming from a well recognized OpenCPN repository -  It is a courtesy build for the boat _FrouFrou_! It is **not** `DashT` but a fork of `tactics_pi v1.0.11`.

JoostFrouFrou, 2020-04-13:
Tnx so far,

Joost

canne, 2020-04-13:
@tom-r  - FYI, the above work is in my v1.0.11 forked repo, commit https://github.com/canne/tactics_pi/commit/df083af146dc1cba2f168dd2efd12230e581dcf0 - here's all the patch files since the fork (all above work): 
[PNKPE-03_patches_since_v1-0-11.zip](https://github.com/canne/dashboard_tactics_pi/files/4469906/PNKPE-03_patches_since_v1-0-11.zip)

Let me know what you think, Maybe test it on your boat (once you get there). I will not make a PR for now: I can understand it is tricky since the meaning of fields changes, not the number. Also, not using fixed VMG angles is something you may want to think about.

What a mess, that NKE!

tom-r, 2020-04-14:
Hi Joost, Petri,
I think there are 3 questions open to be answered:
1.) wrong data definition of $PNKEP, 3 ?
2.) Polar performance shows 100% whereas VMG is still below 100% ?
3.) why does the VMG algorithm use fixed angles 60° upwind, 120° downwind ?

Let's start with the easy one :-)

**ad 1) wrong data definition of $PNKEP, 3 :** 
       I got the $PNKEP definitions from my NKE dealer in Oct. 2015.
       I still have the email with the document attached.
       I think he received it directly from NKE ...
       Either there's a bug in it, or they changed the setup in the meantime ...?
       Anyway, I'm happy to introduce Petri's fix into tactics_pi, thanks for that !!!!

**ad 2) Polar performance shows 100% whereas VMG is still below 100% :** 
            Polar performance always compares the momentary data of boat-TWA/StW against
            the polar-TWA/speed.
            If you change your course 5° more to upwind (TWA gets smaller) the comparison is
            also adopted to the new polar-TWA.
            In the picture below, this is the green/black vectors rotating with your actual TWA.
           "VMG %" and "polar performance %" only match in one spot (assuming you have a 
           "normal" polar), when you sail optimum VMG angle. 

**ad 3) why does the VMG algorithm use fixed angles 60° upwind, 120° downwind :** 
    CMG (optimum speed towards a waypoint) and VMG (optimum speed upwind) basically
    work similar.
    Imagine your polar overlay on the chart : the polar always rotates with the wind
    direction.
[image]

   Graphically, you draw a direct line from your boat to the target (waypoint for CMG), 
   overlay your boat position with the polar centered to wind, and the current CMG angle is 
   determined where a perpendicular line from the waypoint vector touches the polar (tangent).
   See the picture please. This is the pink vector named CMG + the blue tangent.

  **VMG is exactly the same, just in a way that the "waypoint" lies directly in the direction of 
  the wind (upwind or downwind)**
  That's the red VMG vector. Again, we now draw a perpendicular tangent on the waypoint-
   vector (for VMG directly upwind or downwind) and see where it touches the polar.
   That is the TWA and optimum speed you can reach up- or downwind...

  _Calculation wise_, this is fairly easy to determine, it's simply the highest (upwind) / 
  lowest (downwind) polar point in the pic.

   We'll always end up at those 2 specific points for 'opt.VMG.'

   And therefore I just need to know : are we sailing upwind or downwind ?
   So simply 2 fixed values (60/120) will do, to search/iterate the correct half of the polar 
   up/downwards...

Was that understandable ? :-)

BR,
Thomas

canne, 2020-04-14:
Thanks, Thomas, for having taken the time a detailed answer! I always enjoy to read you and your clear analysis. Yes, you are perfectly understandable!

I will split my answer in three different comments according your breakdown to allow quick closing of some of the subjects.

Best regards,
Petri

canne, 2020-04-14:
**ad 1)** There are only two users that I know (me and Joost) who could benefit from this PNKEP,03 patch and Joost got his patched tactics_pi and M. Macron just told me that I cannot see my boat anytime soon, therefore there is no rush:

As I mentioned, there is no way to send two distinct PNKEP,03 messages to satisfy everybody since the content of the message fields changes and not the format of the message.

Are you sure that you want to take this PNKEP,03 message interpretation, which will break the function if there are any of those NKE systems out there your dealer was referring to in 2015?

Think about it, no rush for me (my next update for DashT is months away, let's first see what OpenCPN v5.1 will bring along...).

canne, 2020-04-14:
**ad 2)** For me this is crystal clear, not an issue for me.

canne, 2020-04-14:
**ad 3)** Here our interpretation for the right method is diversifying. Your code contains two, distinct methods for the same thing:
[tactics_pi.cpp#L5161](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/tactics_pi.cpp#L5161)
```
	if (mAWA <= 90)
		tvmg = BoatPolar->Calc_TargetVMG(60, mTWS);
	else
		tvmg = BoatPolar->Calc_TargetVMG(120, mTWS);

	// get Target VMG Angle from Polar
	//tvmg = BoatPolar->Calc_TargetVMG(mTWA, mTWS);
```
[performance.cpp#L234](https://github.com/tom-r/tactics_pi/blob/e97aede98dc203e6ec1f1ce17c6ff19947fe1b09/src/performance.cpp#L234)
```
				else if (m_displaytype == POLARTARGETVMG){
                  TargetxMG targetVMG = BoatPolar->Calc_TargetVMG(mTWA, mTWS);
					if (targetVMG.TargetSpeed > 0) {
						double VMG = BoatPolar->Calc_VMG(mTWA, mSTW);
						double percent = fabs(VMG / targetVMG.TargetSpeed * 100.);
                        targetVMG.TargetSpeed = toUsrSpeed_Plugin(targetVMG.TargetSpeed, g_iDashSpeedUnit);

						m_data = wxString::Format("%d", wxRound(percent)) + _T(" % / ") + wxString::Format("%.2f", targetVMG.TargetSpeed) + _T(" ") + stwunit;
```
Our discussion is about the first snippet, which produces the data to be sent to PNKEP,03 sentence. 

I think it is not necessary to find out if you are going upwind or downwind prior to a call of 

`TargetxMG Polar::Calc_TargetVMG(double TWA, double TWS)`

because your algorithm there makes distinction between upwind and downwind sailing already!

How the current v1.0.11 code proceeds is depicted in the below diagram:
[image]
You assume that the optimal angle is 60 degrees (1) which allows you to find the sweet spot (2) from the polar and then project that to the vector pointing to the wind, determining the polar VMG (3).

What I've done in my patch in GH and in the Joost's version, is that I have applied the second snippet's method (_i.e._ the method of the numerical instrument) in to the separate method which is used to calculate the percentage values of the polar VMG to be sent with PNKEP,03 sentence.

To my understanding, the numerical instrument and the PNKEP,03 sentence should match - at least closely despite the fact that there are two different calculations separated in time.

IMO, the calculation should be the following and I am not sure if your `Calc_TargetVMG()` is doing that since it is late and the code is challenging...
[image]
One should determine a perpendicular tangent across the wind vector and move that down towards the (rotating bug frozen with our calculations speeds) polar until it touches it (1). This sweet spot determines the target VMG angle (2) and the target VMG angle (3) in one go.

JoostFrouFrou, 2020-04-17:
Hi Petri,

Finally got answer from NKE and quess..

[NKE_PNKEP.pdf](https://github.com/canne/dashboard_tactics_pi/files/4492443/NKE_PNKEP.pdf)

Kind regards,

Joost

canne, 2020-04-17:
Hi Joost,
Good to have this confirmation. It looks to me that there might be many versions of NKEs out there and maybe in the past the interpretation was different. After all, it is NKE's own, proprietary sentence, not a NMEA-0183 standard, so they do what they want with it.
I'll let Thomas take the decision, this is his code. There is no hurry. We TL25 owners have got a temporary solution now, thanks to your efforts, Joost!
Best regards,
Petri

canne, 2020-05-17:
Fixed as in the above proto in https://github.com/canne/dashboard_tactics_pi/commit/bf5a20bfdb4312f537f1b7a1afa9945a36730c30 for v1.98.105 - will need to create a regression test and document the change. Closing this long, long thread for now, nevertheless.

commit bf5a20bf (2020-05-16): fix: #110 TL25 NKE compatible 03-sentence
closed 2020-05-17 by canne