canne/dashboard_tactics_pi #113: in two Canvas Opencpn, Dashboard docks in in right Window when moving it.

BerndCirotzki, 2020-05-20:
[image]

[image]

- DashT Version: 1.98.
- OpenCPN Version: 5.1.518
- Platform (Win/Linux/RPI/MacOS + version) : Windows 10

canne, 2020-05-20:
Hi, thanks for reporting,

FYI, I have had that with Dashboard and Tactics already in v5.0, it kind of hoovers it in and docks it even if I do not want it... Something to do with what is called AUI handling (memorizes the location and the docking of windows floating above) in OpenCPN. I noticed that they recognized "Dashboard" plug-in somehow in there but no other plug-ins in any particular way. Maybe it works with "official" Dashboard now, have you tested?

Anyway, without help of the OpenCPN base code there is not much I can do, I am afraid.

But since they are in beta version and have other worries and me in alpha version, let's leave this open and I'll test it again once the OpenCPN v5.2 comes out.

BerndCirotzki, 2020-05-20:
Ok!
... I tested it in the included Dashboard. There it works. But there the "Wind Instrument" looks others.
I liked the old one. (App Wind + Windspeed). But I thing, that's a matter of taste.

thank you.

canne, 2020-05-20:
Yep, my memory was right, I was looking for the OpenCPN code and, indeed they have fixed it for their own "Dashboard" right there, in their code. Here's issue report and the fix https://github.com/OpenCPN/OpenCPN/issues/1544 . I reckon I will be out of luck with this: I cannot impersonate to be "Dashboard" to get special treatment from OpenCPN.

For the record (for myself), their code fix for OpenCPN is [here](https://github.com/OpenCPN/OpenCPN/blob/bcc75f0498748634a16c58be0512ae4aab80d3c5/src/OCPN_AUIManager.cpp#L655) - I will take a look later on if there is a way to do something for this, being a non-official plug-in. No promises.

Regarding the wind instrument in _DashT_: most of the instruments are coming from Tactics not from Dashboard and that's why they are a bit different. Only if an instrument was not in Tactics, I took it from incorporated Dashboard. And here and there I've made small fixes to all instruments.

canne, 2020-05-21:
Created requirement specification for the expected behavior for testing #116 .

canne, 2020-05-23:
For record, [received comment](https://github.com/canne/dashboard_tactics_pi/issues/116#issuecomment-633008718):
> Now OpenCPN crashed direkt while starting up! after uninstall dashboard_tactics, Opencpn works.
> The includes Dashboard is disabled. Sorry :-(

canne, 2020-05-28:
Win Installer for https://github.com/canne/dashboard_tactics_pi/commit/fe87df80f19b9bd246a7f95855fcdfe1d13351b3

[dashboard_tactics_pi-1.98.1061-ov51-win32.exe](https://cloudsmith.io/~canne/repos/DashT/packages/detail/dashboard_tactics_pi-1981061-ov51-win32exe/)

canne, 2020-05-28:
https://canne.testquality.com defect [D34](https://canne.testquality.com/project/7362/result/2018279/defect/6771). Do not remove or edit.

canne, 2020-05-28:
[2020-05-28_tc109_run_tcr849_tpr88_two_pane_hoover_regression.pdf](https://github.com/canne/dashboard_tactics_pi/files/4698298/2020-05-28_tc109_run_tcr849_tpr88_two_pane_hoover_regression.pdf) : PASS.

Closing for now.

commit fe87df80 (2020-05-28): fix: #113, tc109 2 panel Hoover effect regression
closed 2020-05-28 by canne