canne/dashboard_tactics_pi #116: O-DT-DA-1-005 : Support O dual-canvas move and two-edge docking

canne, 2020-05-21:
The product shall provide user a possibility to move the instrument panel from one pane to another in OpenCPN dual-canvas configuration and allow docking either in the middle right edge or the right pane's right edge.

BerndCirotzki, 2020-05-23:
Now OpenCPN crashed direkt while starting up! after uninstall dashboard_tactics, Opencpn works.
The includes Dashboard is disabled. Sorry :-(

canne, 2020-05-23:
Hi, don't be sorry, this is life: beta + alpha = crash is not unexpected since no testing. No worries. If you make new findings in this domain, please continue feed the original issue ticket, #113 you submitted. It would be easier to follow up. Thanks again for reporting!

canne, 2020-05-28:
https://canne.testquality.com requirement [R50](https://canne.testquality.com/project/7362/testcase/242720/requirement/10493). Do not remove or edit.

canne, 2020-05-28:
Successively proven that the function has been accomplished but only regression caused by changes in the base program did not allow to meet this requirement. Regression fix test report in #113 

Requirement met.

Closing.

closed 2020-05-28 by canne