canne/dashboard_tactics_pi #129: Opencpn Download links don't work.

rgleason, 2020-07-03:
## Expected Behavior
Branch removed  1.5.111
## Current Behavior
Not working
## Possible Solution
I sent K alternate URLs to Releases

## Steps/Code to Reproduce the Problem
See https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html#post3177787

## Specifications

- DashT Version: 1.5.111 
- Platform (Win/Linux/RPI/MacOS + version) : All

canne, 2020-07-03:
Hi, recommended links for an OpenCPN users (not developer) are:

Latest stable release: https://github.com/canne/dashboard_tactics_pi/releases/latest

which currently is: https://github.com/canne/dashboard_tactics_pi/releases/tag/v1.5.111

(use "Assets" tab to get the binaries)
<hr />

For eventual developer, project status, the road map and branches explained here: https://github.com/canne/dashboard_tactics_pi/blob/dev200/docs/developers/README.md

If one wants to fork v1.5.111 in view of an eventual pull request: https://github.com/canne/dashboard_tactics_pi/tree/pr151

If the PR gets accepted by me (a bug fix: see the README above), it  will be merged in the maintenance branch: https://github.com/canne/dashboard_tactics_pi/tree/mnt151.

In short, v1.5 minor release is not maintained actively anymore. It is at the level of OpenCPN v5.0 and there is no plan for a new minor version of it. If somebody likes to do a fork and continue its development, please fork `mnt151` branch, of course. Current development is in `dev200` branch. Once finished, it will merged into  `master` branch which currently is not active (_i.e._ only one developer working). When this happens? I reckon we will be close to the New Year by then. But which one...

canne, 2020-07-03:
A teaser from [dev200bstrap](https://github.com/canne/dashboard_tactics_pi/tree/dev200bstrap) branch:
[image]

rgleason, 2020-07-04:
Race Start like Steve had done. Nice.

canne, 2020-07-18:
Looks OK now on OpenCPN web site. Closing.

closed 2020-07-18 by canne