canne/dashboard_tactics_pi #130: Update to  JonGough's Frontend2  1.0.124  for Opencpn Plugin Manager?

rgleason, 2020-07-03:
Dear Petri,

We are getting close to having pretty complete and reliable system for Plugin Manager that builds and deploys 13 environments which get added to the Plugin Manager master catalog.

Will you consider updating your frontend to that system at some point when Jon Gough advises it is ready ?
He has just resolved some language issues that came up.

https://github.com/jongough/testplugin_pi

canne, 2020-07-03:
Dear  Rick,
see answer https://github.com/canne/dashboard_tactics_pi/issues/129#issuecomment-653616167 - for this one the answer is "no, I am not interested to do that work anymore for v1.5.1 where it would be, indeed applicable". For current development in `dev200` the answer is "yes, I am interested in it, but further build automation and OpenCPN distribution is currently not in my agenda". I have CI and some automated testing for Windows, MacOS and Ubuntu 20.04 and it is largely sufficient. For Rasberrry, with no wxWebView available natively, I build on  my RPI. My setup is the one of a developer, not that of a distributor, of course.

If this is an issue, one needs to find somebody who wants to take over `mnt151` branch. I do not have time for that since working on other, quite demanding things right now.
Best regards,
Petri

P.S. on those "13 platforms", is the wxWebView working out of the box? That would be interesting information (read [here](https://github.com/canne/dashboard_tactics_pi/releases/tag/v1.98.104) about my findings on *nix platforms)

rgleason, 2020-07-04:
Leamas answered your question here
https://github.com/OpenCPN/OpenCPN/issues/1966#issuecomment-653728096

canne, 2020-07-04:
Thanks, Rick, for getting the clarification to my question this so quickly, I am impressed! Leamas' answer confirms my observations. FYI, In OpenCPN 5.2 they would use wxWebView widgets only for help and some other, maybe (future) web based services they plan. Not a big deal for OpenCPN to strip those features in build time - it is a quite natural way to support older platforms: strip features.

Back to my road map and how it may affect your road map if you plan to provide Plugin Manager support for DashtT:

**v1.5.1**11 and any derivation from that ; no problem since it has no wxWebView dependencies whatsoever. The only problem would be to find a maintainer for the big baby...

**v2.0** and onward : it has strong wxWebView dependencies and I do not have any plans for a "feature stripped version" of this mainly performance oriented tool:. Therefore Plugin Manager build process and also the distribution process should accept or find a workaround on older (GTK2) target that there is a plug-in of which all dependencies cannot be fulfilled by the native OS distribution. Otherwise, I will keep supporting only GTK3 based systems on *nix using my well established build system.

Good news for RPI is that there is a RPIOS/64 beta now out. Quite likely, it is a fork of Ubuntu 20.04 LTS or such. Not sure but it would be natural that it would be GTK3 based and thus DashT v2.0 would work on it out of the box.

canne, 2020-07-18:
https://canne.testquality.com requirement [R56](https://canne.testquality.com/project/7362/testcase/250242/requirement/10591). Do not remove or edit.

canne, 2020-11-21:
Because of many dependencies to the outside world services, cannot be done as per today without these features in OpenCPN:

1. https://opencpn.org/flyspray/index.php?do=details&task_id=2710 ;
2. https://opencpn.org/flyspray/index.php?do=details&task_id=2715 ;
3. GTK2 is the default for OpenCPN if GTK2/GTK3 madness system ;
4. wxWidgets 3.1.4 or superior is not used in Linux but what is available, 3.0.4 usually.

Maybe after a few years all the above issues have been resolved by natural development of run-time platforms, OpenCPN, users and even those third party services like databases. Meanwhile the few potential users need to find their way with the documentation, which is not that bad. There is no point either to start making degraded versions of this code base for somebody who just needs many crazy jumping dials.

Resolving as Won't fix.

closed 2020-11-21 by canne