canne/dashboard_tactics_pi #160: Can't get Signal K data without using NMEA0183 / "- No template." message

romton843, 2021-01-05:
## Security Priority

  - [ ] This is a report related to [Security](github.com/canne/dashboard_tactics_ptyi/security)

## Installation checklist (change to [X] when sure)

I have installed the software for which I am reporting the issues from:

  - [ ] This repo's [Releases](https://github.com/canne/dashboard_tactics_pi/releases)
  - [ ] Your own fork and build [Link](https://github.com/USER/REPO)
  - [X] OpenCPN's plug-in manager
    - [ ] Plug-in manager issue reported in [CF DashT thread](https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html)
  
## Specifications

- OpenCPN Version: 5.2.4
- DashT Version: 1.6
- Operating System :
  - [ ] Windows
  - [X] Linux
    - [ ] Ubuntu
    - [X] RPI
    - [ ] Debian
    - [ ] Fedora
    - [ ] Other: ____
- Operating System Version : Raspbian GNU/Linux 10 (buster)

## Expected Behavior

- [X] As described in the [doc](https://dashboard-tactics-pi.readthedocs.io/en/BRANCH/CHAPTER.html#SECTION) ( [example](https://dashboard-tactics-pi.readthedocs.io/en/dev200/enginedjg/enginedjg.html#Subscribe-to-data) )

Hi, the only way I can get Signal K data to be available in OpenCPN is by using the Signal K to NMEA0183 plugin and a NMEA0183 connection. However I thought the data should be available with simply a Signal K connection in Open CPN.
My Signal K connection settings are:
Network, Input, Signal K, localhost, 3000, 1

## Current Behavior

- [X] I've been following [troubleshooting](https://dashboard-tactics-pi.readthedocs.io/en/BRANCH/CHAPTER.html#SECTION) ( [example](https://dashboard-tactics-pi.readthedocs.io/en/dev200/enginedjg/enginedjg.html#Troubleshooting) )

In your instrument panel, the Signal K In instrument shows "- No template." and nothing else and when I try to add an instrument with Signal K data it shows "HALT".

## Possible Solution

## Steps/Code to Reproduce the Problem

1. step1
1. ...

```
// ... code or configuration
```

SCREENSHOTS, (zipped) log files, VDR recordings or Signal K logs.

canne, 2021-01-05:
Hi, starting from OpenCPN v5.2.0 you do not need anymore DashT just to get Signal K data - it is supported natively in the Connections dialog:
[image]
See https://opencpn.org/OpenCPN/about/ver520.html

If this is what want, you do not need DashT and can stop reading now.

If the above does not work, you will get the best help from [Cruisers Forum](https://www.cruisersforum.com/forums/f134).

Supposing that you _want_ to use DashT and Signal K data with it, be aware that it has an independent network connection, [directly to Signal K server node, not going through the OpenCPN](https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/signalk/signalk.html#Signal-K-data-interchange). Why this would give you the "_No template_" message?

This is because you use, as you have indicated OpenCPN Plug-in Manager: it fetches, from its own data repositories a fork of this (original) DashT. Unfortunately, I cannot but guess what goes wrong but I have an idea:
>the Plug-In Manager has changed the data structures for the 'managed' plug-ins. In the OpenCPN fork of DashT this has not been taken into account, I guess: At the first start, _SignalK In_ "instrument" is looking for a template configuration file from the "non-managed" plug-in's data directory, to be copied to the working directory but it cannot find nothing in that location.

If you still want to try with DashT Signal K In "instrument" on a RPI, your may want to report about the above issue in the [Cruisers Forum thread mentioned above](https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html), maybe there is somebody willing to do a patch with the correct file paths in the "managed" DashT.

I am sorry that I cannot provide you a solution - the current DashT version v2.0.1 does not support RPI in any of is native OS - this is due to the lack of GTK3 which is mandatory for v2.0 onwards. But since I have a couple of RPIs myself, I have published in the [documentation instructions how the DashT version available here can be made running on a RPI with some restrictions](https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/installation.html#wxWebView-and-GTK2). On RPI, I am running Ubuntu Focal Fossa 20.04 LTS and DashT v2.0.1.

closed 2021-01-05 by canne

canne, 2021-01-05:
Oh, furthermore, so that you do not need to waste your valuable time, only the version v2.0 of DashT understand _both_  Signal K native data format and NMEA-0183 format. The earlier versions (on which the OpenCPN Plug-In Manager forks are based) understand only NMEA-0183. Also, in those earlier versions you do not have any web-based instruments, like the EngineDJG dials.

If this is what you are looking for (EngineDJG + native Signal K data on a native RPI OS Buster 32-bit), your only alternative is to follow the above instructions. I have [tested it in the past with some beta versions](https://github.com/canne/dashboard_tactics_pi/releases/tag/v1.98.104), it works (there are still some binaries). But I will not support it since it is an inherently obsolete platform with is GTK2 based libraries and as such a dead end for a developer....

romton843, 2021-01-06:
Hi, thank you for your fast answer, I’m sorry I thought your plugin was the one providing OpenCPN with Signal K data. I understand better now. I guess I’ll try to solve my issue with the Cruisers Forum. Best regards.
Romain