canne/dashboard_tactics_pi #166: Question: About  branch pr151+mnt151-frontend2 on my repository

rgleason, 2021-11-08:
Dear Petri 

I have sort of lost track of this branch which was used for release v1.6.5.0
https://github.com/rgleason/dashboard_tactics_pi/tree/pr151%2Bmnt151%2Bfe2
which ended with  v1.6.5.0 +fe2 v1.0.132 + mnt151. 
 **Your last commit was  feat: v2.0.0-alpha-08 bait  August 12,2020**
There are many of your commits from May through August including "feat"

The most recent release v1.6.7.0 used my mnt151-frontend2 branch 
https://github.com/rgleason/dashboard_tactics_pi/commits/mnt151-frontend2.  
 **Your last commit was for an edit to "docs" May 16,2020.**   
On August 24 we released v1.6.6.0.  
May 16, 2020  rasbats added the asciidoc/antora "manual" 
On Nov 7, 2021 I released v1.6.7.0 from this branch.

**Which branch should we be using?**
I hope you can straighten us out on that!

Recently, using this new release v1.6.7.0 I have been fielding a question about why true and apparent wind don't correspond when SOG=0.  here  https://www.cruisersforum.com/forums/f134/dashboard-tactics-dash-t-226601.html#post3516766

canne, 2021-11-08:
Dear Rick,

I am not working in this project anymore, it has met its goals. The code base remains for community to use.

Unfortunately I have no way to tell how your repository has evolved since you made a fork . My repository and its milestones tags are registered here: https://github.com/canne/dashboard_tactics_pi/blob/dev220/docs/developers/README.md

If you scroll down, there is a diagram illustrating all significant branches and tags. Referring to individual commits is a mission impossible... You can see that there is a last tag v1.5.111 in branch mnt151.

All other tags and branches  is DashT v2.0 development which is, I presume, incompatible with your v1.6.7.0 or such.

Regarding your SOG=0 question, I presume this is the same as the requirement https://github.com/canne/dashboard_tactics_pi/issues/84 which I can see that I have implemented and tested in the 19th of October, 2019, in https://github.com/canne/dashboard_tactics_pi/issues/85. Please take a look, maybe that helps, there seems to be a [test report](https://github.com/canne/dashboard_tactics_pi/files/3748202/2019-10-20_TCR817_TC108_Test_TWD_calc_standstill_test_report.pdf) which you can compare to your own use case, perhaps.

Best regards,

Petri

closed 2021-11-08 by canne

rgleason, 2021-11-08:
Dear Petri, thanks for the quick and very thorough response.

I have pointed to #84 on Cruisers Forum and will do so again for Hans. Thank you for such great documentation. It truely is a marvel!  We are all so lucky to have had you working on this.

This diagram is a great help https://github.com/canne/dashboard_tactics_pi/blob/dev220/docs/developers/v2.0.0_holdpoint_threads.png

Yes this last tag  [v1.5.111 in branch mnt151.](https://github.com/rgleason/dashboard_tactics_pi/commits/mnt151-frontend2)  is in my mnt151-frontend2 thread which we are using currently for v1.6.7.0  so we have this right.

I understand there is someone building the master branch v2.20 from your repository and running it in windows [Hans]
Do you think I should I be trying to build this version now?

Best regards,
Rick

canne, 2021-11-08:
> I understand there is someone building the master branch v2.20 from your repository and running it in windows [Hans] Do you think I should I be trying to build this version now?

If you can build from and can target [v2.0.3](https://github.com/canne/dashboard_tactics_pi/releases/tag/v2.0.3) exclusively on Windows 10 or greater and Ubuntu 20.04 LTS platforms (GTK3 and wxWebView)  while keeping other, GTK2 platforms w/ your v1.5.111 deviated distribution, it is doable and would match the documentation. I do not have time to assist in that endeavor, I am sorry.

Best regards,
Petri

rgleason, 2021-11-08:
Petri, I can do that and will do so. I will check grafana and webview for flatpak support too and if available will build that too for gtk3.

Thanks so much